genodelabs/genode #2521: Warning: blocking canceled in entrypoint constructor

senier, 2017-09-13:
When I run Genode/base-linux on TravisCI, I get the following error since the last update of their environment:

```
genode build completed
using 'core-linux' as 'core'
using 'ld-linux.lib.so' as 'ld.lib.so'
using 'linux_timer_drv' as 'timer'
spawn ./core

Genode 17.08
17592186044415 MiB RAM and 8998 caps assigned to init
Warning: blocking canceled in entrypoint constructor

Error: Test execution timed out

make: *** [run/init] Error 254
make: Leaving directory `/home/travis/build/Componolit/genode/build'
```

Unfortunately, I cannot reproduce the issue on any system I have. The error happens with 17.05 as well as 17.08. When I downgrade the TravisCI environment to their previous version (deprecated-2017Q3) the test works as expected.

The most notable difference between the two environments is a different kernel package:

```patch
-Build group: deprecated-2017Q3
+Build group: stable
 Build dist: trusty
-Build id: 275180532
-Job id: 275180533
-Runtime kernel version: 4.4.0-83-generic
+Build id: 275171177
+Job id: 275171178
+Runtime kernel version: 4.4.0-93-generic
```

Both environments are based on Ubuntu 14.04.5 LTS. See https://github.com/Componolit/genode/tree/travisci_blocking_canceled for a reproducer.

senier, 2017-09-14:
I just realized that downgrading the environment does not solve the problem. The reproducer referenced above still fails sometimes, butmore complex scenarios always fail. Looks like a race condition.

cproc, 2017-09-14:
We could reproduce the problem on Ubuntu 16.04 with the 4.4.0-93-generic kernel, but not with other kernel versions so far. There have been changes in this specific kernel version regarding the way how the dynamic linker is loaded by the Linux kernel ([1][2]), which seem to be related to the problem.

When your tests fail in the downgraded environment, is it still the same log output?

[1] https://github.com/torvalds/linux/commit/eab09532d40090698b05a07c1c87f39fdbc5fab5
[2] https://launchpad.net/ubuntu/+source/linux/4.4.0-93.116

senier, 2017-09-14:
I diffed the logs of two otherwise identical builds on stable and deprecated and there are no notable differences. I'm trying some test cases more complex than `run/init` now.

chelmuth, 2017-09-15:
@cproc was referring to your statement in https://github.com/genodelabs/genode/issues/2521#issuecomment-329373314: *I just realized that downgrading the environment does not solve the problem.* We couldn't reproduce the faulty execution of *ld.lib.so* on other Linux kernels than 4.4.0-93-generic. The Genode log itself is not helpful as our investigation uncovered that the execve for ld.lib.so fails *in the Linux kernel*. Currently, we're identifying options to simplify the binary layout of the linker to fit into the narrow space of *only two ELF segments which must be **text** followed by **data** which itself must end with the **BSS section***.

chelmuth, 2017-09-18:
Thanks to @ssumpf's efforts I generated 19dbed0 which fixes the issue for me on Ubuntu 16.04.2 LTS with 4.4.0-93-generic.

senier, 2017-09-18:
Thanks to @ssumpf and the others for addressing that issue! I can confirm that base-linux works again in the latest and the previous TravisCI environment.

jklmnn, 2017-10-04:
I can also confirm that 03a17e2cf61879cdfecb32df1484b3d058231a48 fixed that issue for me on Debian with 4.9.0-3-amd64.

commit 19dbed0d (2017-09-18): linux: stack-area handling with recent Linux kernels

We moved the stack-area segment 128 MiB behind text and data to comply
with assumptions in the kernel ELF loader.

This commit also reenables static binaries on linux and removes the
unused stack_area.stdlib.ld script.

Fixes #2521
closed 2017-10-19 by chelmuth