genodelabs/genode #2525: usb_drv: could not revert session RAM quota (service=Platform) when restarting

BorisMulder-CSL, 2017-09-22:
I have the following scenario:
init has a child platform_drv and another child init (named init_1). init_1 has another init child named init_2. init_2 has a usb_drv as a child. In short, there are two components that are important and some inits:
- [init -> platform_drv]
- [init -> init_1 -> init_2 -> usb_drv]

Now I empty and fill the config of init_2 in order to restart all children, including usb_drv.
When I empty it by removing all start nodes (and kill usb_drv), I get the following warning:
```
[init_1] resource_request: ram_quota=6331631
[init_1] resource_request: ram_quota=6331631
[init_1] Warning: init_2: could not revert session RAM quota (service=Platform cid=188 args='cap_quota=2, label="init_2 -> usb_drv -> ", ram_quota=14814, diag=0' state=CLOSED ram_quota=6330846, cap_quota=2)
```

Now when I fill the config again with start nodes (restarting usb_drv), I get the following warning:
```
Warning: PD (init -> platform_drv) RAM limit (used=13096K, limit=13488578) exceeded during transfer_quota(192K)
[init -> platform_drv] resource_request: ram_quota=192K
[init] child "platform_drv" requests resources: ram_quota=192K
```

platform_drv crashes because it is out of RAM and usb_drv of course hangs waiting for platform_drv.
This looks like a bug in either init, platform_drv or usb_drv. On top of init_2 other components are run and restarted in this way that use the platform session from init -> platform_drv (such as ps2_drv and fb_drv). They do not give these problems; so it seems to me the problem is in usb_drv.

Any idea what could be the issue and how to fix it?

BorisMulder-CSL, 2017-09-22:
Edit: I can also see that if usb_drv is moved to init_1 instead of init_2, any client connection (in my case that would be one from input_merger) are not released, preventing the restarted input_merger from reconnecting with usb_drv. When moving the attached usb mouse in this case this warning is spewed out many times:
`[init_1 -> usb_drv] Warning: submitting signal failed - error 4 - context=cap<34280>`

BorisMulder-CSL, 2017-09-25:
update: platform_drv simply does not seem to support reconnecting clients. When I restart the subsystem multiple times, it will always crash with:
`Warning: PD (init -> platform_drv) RAM limit (used=16176K, limit=16593175) exceeded during transfer_quota(192K)`

alex-ab, 2017-11-06:
Should be addressed by #2564, please check.

closed 2017-11-09 by alex-ab