genodelabs/genode #2526: Use grub2 for iso/usb boot on x86

alex-ab, 2017-09-25:


nfeske, 2017-09-25:
Very cool! Until now, I was not aware that the use of GRUB2 alleviates the need for isolinux as well. That is another plus.

chelmuth, 2017-09-29:
From an offline discussion: The patch above fixes this issue but will be reworked by fetching the grub2 binaries via our port mechanism.

alex-ab, 2017-10-04:
64ecf6d8356f94e0720f8d7840986ce442ce67b3 contains a proposal to use our port mechanism to fetch the grub2 binaries

nfeske, 2017-10-23:
Thank you for the fixup, which is exactly what I had in mind!

m-stein, 2017-11-10:
@alex-ab @chelmuth What do you think about this little addition:
dabf9ec tool/run/grub2: user-friendly missing message

It allows the user to simply copy/paste the instruction that prepares grub2 and makes the message conform to the according messages of the common build system.

chelmuth, 2017-11-11:
thanks, merged it to staging.

commit 5343bae4 (2017-11-09): tool: use grub2 instead of grub1 for iso boot

Fixes #2526
commit 16c520f9 (2017-11-15): tool: use grub2 instead of grub1 for iso boot

Fixes #2526
commit 5f1815e1 (2017-11-24): tool: use grub2 instead of grub1 for iso boot

Fixes #2526
commit 44adc3c4 (2017-11-30): tool: use grub2 instead of grub1 for iso boot

Fixes #2526
closed 2017-11-30 by chelmuth