genodelabs/genode #2529: chroot: move from genode-world to genode, add explicit path policy

nfeske, 2017-09-26:
I'd like to incorporate the `chroot` component into the upcoming sculpt scenario (successor of Turmvilla), for example to limit virtualbox' view on the file system to the single directory where the vbox and vdi files are located. By enforcing this sub view in a dedicated chroot component, we can keep the policy of the underlying file-system server agnostic with regard to the varying use cases of the scenarios executed on top.

Currently, the chroot component is part of the genode-world repository. In order to integrate it into the sculpt scenario (which will eventually become part of repos/gems), I would like to have it located in repos/os. Furthermore, it would be nice to be able to specify a `path` attribute to a policy instead of implicitly using the client label to select the path. This would make it very intuitive to use.

ehmry, 2017-09-26:
Would you like me to move it over? Would it be okay to use an implicit path when ``path=""`` is not present in a policy?

ehmry, 2017-09-26:
Fixed at 529d8c4de7bc11a18a2552245cb9d3d0105f8292.

nfeske, 2017-09-27:
Wow that was quick! Thank you. This is perfect. :-)

nfeske, 2017-09-27:
Merged to staging.

commit 88f7ed70 (2017-09-27): import chroot from world repository

Fix #2529
closed 2017-10-19 by chelmuth