genodelabs/genode #2532: signal: fix starvation by fast signal contexts

m-stein, 2017-09-29:


chelmuth, 2017-10-19:
Fixed on master by b811ef4331386fe4601fd751d2a826b0365a28d8

skalk, 2017-11-06:
When bisecting to find the commit that introduced the regression that noux_tool_chain_auto does not work on several architectures anymore, I struck commit b811ef4
@m-stein @chelmuth any ideas?

chelmuth, 2017-11-06:
Sorry, I've no hint, but think it's time to reopen the issue.

m-stein, 2017-11-07:
@chelmuth A bug fix for the framework:
4b875fd signal: fix bug in context list by using a ring

skalk, 2017-11-08:
@m-stein, @chelmuth  : it seems to fix the problem, at least on hw/x86_64 noux_tool_chain_auto runs again with the patch

m-stein, 2017-11-20:
@chelmuth I tested this with ``timeout timer signal nic_router`` on several machines and kernels so I hope it doesn't break too much: 
c5458cf fixup "signal: fix bug in context list by using a ring" (for-each lambda)
6b0fcb4 fixup "signal: fix bug in context list by using a ring" (head is next)
1415b6c fixup "signal: fix bug in context list by using a ring" (naming)
57652ef fixup "signal: fix bug in context list by using a ring" (naming)

skalk, 2017-11-29:
To me it seems the issue is now finally fixed, isn't it?

m-stein, 2017-11-29:
Yep.

m-stein, 2017-12-05:
@chelmuth I created a fix in agreement with @alex-ab:
4c1c2c5 signal: fix race in Signal_receiver pending_signal

commit decd303a (2017-12-05): signal: fix race in Signal_receiver pending_signal

In the past, we had the lock of the context we found for the rest of the scope
of pending_signal. Now we use for_each_locked to find the context and the
lock is released as soon as we return from for_each_locked. Thus, we need to
lock the context again before returning it to avoid that the copy constructor
during the return reads values that are currently changing.

Fixes #2532
commit b812179e (2017-12-21): signal: fix race in Signal_receiver pending_signal

In the past, we had the lock of the context we found for the rest of the scope
of pending_signal. Now we use for_each_locked to find the context and the
lock is released as soon as we return from for_each_locked. Thus, we need to
lock the context again before returning it to avoid that the copy constructor
during the return reads values that are currently changing.

Fixes #2532
closed 2017-12-21 by chelmuth