genodelabs/genode #2533: VFS server hangs when attempting to open a directory as file

nfeske, 2017-09-29:
I have integrated the VFS server along with the Rump VFS plugin into the sculpt scenario (https://github.com/nfeske/genode/commits/sculpt_debug) and observe a strange behavior when combining it with fs_rom: On disk, I have a directory called `/config`. While directing ROM session requests of a sub system (i.e., the one in the wm pkg archive) to fs_rom, fs_rom happens to receive a request for `wm -> layouter -> config`. Taking the last element ("config") as file name, it tries to open the file `/config` at the VFS server. The scenario blocks here and no other VFS client is able to interact with the VFS. In contrast, when accessing a non-existing file, the VFS server handles the situation gracefully.

nfeske, 2018-01-29:
Possibly related: We got a report that `cat /bin` gets stuck spinning when executed within Noux. This can be reproduced with the noux_bash.run script.

ehmry, 2018-01-29:
I will look into it.

ehmry, 2018-01-30:
@nfeske Is it still necessary to allow directories to opened as files at the VFS after adding the `opendir` method? https://github.com/genodelabs/genode/blob/601388902822e985cc7189876e4f38844879a231/repos/os/include/vfs/dir_file_system.h#L542

Looking into the Libc and Noux I think there should be additional checks to prevent reading directories as files. At least on Linux I get EISDIR when I try to do this.

nfeske, 2018-02-10:
API-design wise, I cannot give you a definite answer to the first question. Let's save this for later.

But addressing the latter point in the libc would fix this issue.

There is also another - related - issue:

When typing just `/` in noux, we get an error message "Error: binary is not an ELF" and noux hangs.

ehmry, 2018-02-13:
Fixed on my **issue2533_dir** branch.

nfeske, 2018-02-13:
That's very nice - immediately merged to staging.

commit 27a2a829 (2018-02-13): Vfs::Dir_file_system: do not use Vfs_handle for opening directories as files

The Dir_file_system uses static cast to convert handles from the
application to a plugin local type. For this reason, only the local
handle type may be returned from 'opendir' or 'open'. This fixes the
unexpected behavior when opening directories as files.

Fix #2533
commit 013eb506 (2018-02-15): Vfs::Dir_file_system: do not use Vfs_handle for opening directories as files

The Dir_file_system uses static cast to convert handles from the
application to a plugin local type. For this reason, only the local
handle type may be returned from 'opendir' or 'open'. This fixes the
unexpected behavior when opening directories as files.

Fix #2533
closed 2018-02-16 by nfeske