genodelabs/genode #2534: nic router: dhcp client functionality

m-stein, 2017-10-04:


m-stein, 2017-10-16:
@chelmuth These commits should solve the issue:
4f9c8fe nic_router: rename and move Ip_allocation
af4824a nic_router: move Dhcp_server to extra header/unit
ae70209 virtualbox_nic_router: two routers, all dynamic
efd3b94 nic_router: minor style fixes
98f5d0d nic_router: DHCP client functionality
015745a nic_router: encapsulate IPv4 peer config in class
d82389c net: move Ipv4_address_prefix to nic_router
42b1bb3 dhcp: support reading of some further option types

chelmuth, 2017-11-01:
I merged all the commits above but ae70209 as I want your acknowledgment that this change is also intended for staging.

m-stein, 2017-11-02:
@chelmuth Thanks. Yes from my side it was intended to get ae70209 to staging. Is there a particular reason why you ask?

chelmuth, 2017-11-02:
The reason is that I want to prevent merging of debugging artifacts ;-)

chelmuth, 2017-11-06:
The commits above are on master. May we close the issue?

m-stein, 2017-11-06:
Yes.

m-stein, 2017-12-04:
I reopen the issue as the DHCP client functionality has raised some problems in real life scenarios.

m-stein, 2017-12-04:
@chelmuth Please consider merging:
7f2ac9b nic_router: domain-state-verbose flag
5118629 nic_router: do not route to domains w/o IP config

m-stein, 2017-12-06:
Two adaptions to the DHCP-server functionality:
a979a42 nic_router: no DHCP fail on unexpected DISCOVER
f4510fd nic_router: better warnings on bad DHCP requests

chelmuth, 2017-12-06:
Would you mind rebasing your branch to current staging?

m-stein, 2017-12-06:
@chelmuth Yes, done.

chelmuth, 2017-12-06:
Ehm, thanks anyway ;-)

MartijnVerschoor, 2017-12-08:
@m-stein, we still experience the bad dhcp requests https://github.com/genodelabs/genode/issues/2373#issuecomment-332763953 in our scenario. We built on top of current staging. With the increased verbosity the warning now reads:

```
Warning: (vm1) DHCP REQUEST from client without offered/acked IP
```

We are running a Windows 8.1 guest. With respect to the network configuration, there is little difference between your run scripts and ours. Ubuntu 16 guest succeeds on the first DHCP request with the same configuration.

Before we dive in too deep, do you have any pointers or ideas what may cause this (or how to fix it / where to look, etc..). Could this be exclusive for Windows 8.1 guests?

chelmuth, 2017-12-08:
It may be good to know which packets were seen by *nic_router* up to this point. Please enable `verbose="yes"` in the router config.

m-stein, 2017-12-10:
@MartijnVerschoor Do you already use the newest commits for the NIC router since DHCP client functionality?

82bb7ab nic_router: no DHCP fail on unexpected DISCOVER
0b52bff nic_router: better warnings on bad DHCP requests
d2e9723 nic_router: domain-state-verbose flag
0931adb nic_router: add config schema
22fd703 nic_router: do not route to domains w/o IP config
d2bb51a nic_router: advanced timeout configuration
564e6a6 nic_router: read seconds attributes generic
b2ea164 nic_router: fix bugs in DHCP client functionality
9d84d8b nic_router: rename and move Ip_allocation
127ceac nic_router: move Dhcp_server to extra header/unit
5e227f9 nic_router: minor style fixes

At least they should give you a more descriptive warning on "bad DHCP messages". If this doesn't help, indeed, a log with verbose="yes" would be useful.

m-stein, 2017-12-10:
PS: Please be aware that d2bb51a makes the rtt_sec attribute obsolete while d2bb51a and d2e9723 introduce new attributes.

BorisMulder-CSL, 2017-12-11:
> Do you already use the newest commits for the NIC router since DHCP client functionality?

Yes, we are up to date with current staging.

Our verbose log output gives the following:

<details> 
  <summary> Log output for nic_router </summary>
```
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.230.148 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.230.148 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.170.212 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.170.212 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.108.112 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.108.112 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.76.49 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.76.49 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.136.38 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.136.38 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.99.117 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.99.117 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.212.43 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.212.43 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.127.176 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.127.176 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.79.108 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.79.108 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.184.9 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.184.9 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP 02:02:02:02:02:02 > 0.0.0.0 cmd 1
[init_1 -> init_2 -> nic_router] Warning: (vm1) Drop packet: DHCP REQUEST from client without offered/acked IP
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.184.9 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.184.9 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP 02:02:02:02:02:02 > 0.0.0.0 cmd 1
[init_1 -> init_2 -> nic_router] Warning: (vm1) Drop packet: DHCP REQUEST from client without offered/acked IP
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP 02:02:02:02:02:02 > 0.0.0.0 cmd 1
[init_1 -> init_2 -> nic_router] Warning: (vm1) Drop packet: DHCP REQUEST from client without offered/acked IP
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff ARP 02:02:02:02:02:02 0.0.0.0 > 00:00:00:00:00:00 169.254.184.9 cmd 1
[init_1 -> init_2 -> nic_router] (vm1 <- router) ETH 12:34:56:78:90:02 > 02:02:02:02:02:02 ARP 12:34:56:78:90:02 169.254.184.9 > 02:02:02:02:02:02 0.0.0.0 cmd 2
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP 02:02:02:02:02:02 > 0.0.0.0 cmd 1
[init_1 -> init_2 -> nic_router] Warning: (vm1) Drop packet: DHCP REQUEST from client without offered/acked IP
[init_1 -> init_2 -> nic_router] (router <- vm1) ETH 02:02:02:02:02:02 > 01:80:c2:00:00:0e EtherType 0x88cc not supported
[init_1 -> init_2 -> nic_router] unknown network layer protocol
[... cut ...]
[init_1 -> init_2 -> nic_router] New UDP client link: src 192.168.1.2:54837 dst 224.0.0.252:5355 at "vm1"
[init_1 -> init_2 -> nic_router] New UDP server link: src 224.0.0.252:5355 dst 10.0.0.231:49232 at uplink
```
</details>

One thing you will notice is that in the beginning the windows VM is issuing APIPA ARP requests (https://wiki.wireshark.org/APIPA).
BTW I added some print statements to show the EtherType of unsupported network layer protocols in hex (you can see that in "EtherType 0x88cc not supported").
After this the vm seems to be connected fine.

m-stein, 2017-12-11:
@BorisMulder-CSL Thanks for your hint about APIPA! I didn't know about this and - independent from your issue - the referenced article may indicate that we should check the DHCP server config for Special-Use IPv4 Addresses [1] as "They should not be [...] assigned using DHCP". What puzzles me is that APIPA should be used only "If a network client fails to get an IP address using DHCP". But the log contains no initial attempt by the VM to use DHCP. I also did not find any other exception in which the client may directly call DHCP REQUEST without having received a DHCP OFFER. So I wonder whether the VM is trying to do DHCP DISCOVER before its connection to the NIC router is established? This may be the case if there are other NIC-session-intercepting components between the VM and the router. 

[1] http://www.ietf.org/rfc/rfc3330.txt

BorisMulder-CSL, 2017-12-18:
@m-stein often I notice that the VM does a dhcp request before switching to APIPA, but not always. An example is this one:
[init -> nic_router] (router <- vm1) ETH 02:02:02:02:02:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP 02:02:02:02:02:00 > 0.0.0.0 cmd 1
[init -> nic_router] Warning: (vm1) Drop packet: DHCP REQUEST from client without offered/acked IP
I do not have any other components in between vbox and the router.

m-stein, 2017-12-18:
@BorisMulder-CSL With https://github.com/m-stein/genode/commit/8714df47bb5b275040ee35c7817fa52ae1ac3ee9 from my nic_router_configurable_packet_dump branch we could get a deeper insight into your DHCP traffic. It would be interesting to know whether the routers DHCP server is causing the Guest to start with APIPA or if the Guest does APIPA regardless of the router. So, maybe you could post the more verbose output from a case where DHCP is done before APIPA. Another thing to try would be to replace the nic_router in your scenario with a nic_dump and see whether the Guest still does APIPA.

closed 2017-12-21 by nfeske