genodelabs/genode #2535: remove libc socket plugins

ehmry, 2017-10-09:
Bottlenecking all socket support to vfs_lxip makes testing socket_fs and merging vfs_lwip easier.

chelmuth, 2017-10-19:
I merged the series. Please keep in mind that I had a merge conflict in *repos/ports/run/netperf_lwip_usb30.run* which no longer exists due to the changes in ddb2805f3a357ddd24dead379d44db736129da1c. I'll inform you about the nightly results.

m-stein, 2017-10-19:
@ehmry I saw that you changed the test servers/clients in the nic_dump/router tests so that they all use LxIP, which makes them more RAM-intensive (12M -> 32M). Unfortunately, I had to achieve the exact opposite these days because the nic_router test fails on some platforms with insufficient RAM (my branch https://github.com/m-stein/genode/tree/merge_to_staging). I moved all tests to LwIP only and got stuck now while rebasing. Do you see an easy way out of this?

ehmry, 2017-10-20:
@m-stein I will try to get lwip at parity with lxip tomorrow.

m-stein, 2017-10-20:
@chelmuth just explained to me the plan of the LwIP changes. So, I'll simply wait for the next steps. No need to hurry ;)

ehmry, 2017-10-23:
@m-stein my `issue2335_lwip` branch completes `run/nic_router` ok on QEMU x86_64, but not yet on real hardware, I haven't tested ARM.

ehmry, 2017-10-26:
@chelmuth If I may treat this as a general socket_fs issues, I have two small patches 9432033a9e832f45900beecf0cd8217296ef057f and 720c805969875a7b1ee49e49677bbf137c19554b to add to staging.

chelmuth, 2017-12-19:
I merged the commits with some fixups into https://github.com/chelmuth/genode/commits/issue2335_lwip. @ehmry please review and consider the comments I made for various commits. There are also the following open issues / alternative commits I need your help with:

- libc_lxip has to be removed
- nic_dump and nic_router run scripts have to be adapted
- http_srv_static test should be merged into http_srv (keeping configurability)
- http_srv_connect test tool can be removed
- tests adapted from lwip to libc API should be cleaned from references to lwip
- f050ce9 libc: internalize socket support as socket_fs
- f43fb55 lib/vfs/lxip: poll all open handles continously
- 6fff45e libc: default socket_fs path to '/socket'

chelmuth, 2017-12-19:
The comment above is also relevant for #2335 at least.

ehmry, 2017-12-19:
I would like to get LwIP merged before removing libc_lxip because for me at least, vfs_lwip is known to work. The biggest problem with merging LwIP that I can see is removing libc_lwip. Perhaps an intermediate step could be to leave the current LwIP library as it is and introduce a static *lwip_async* or *lwip2* library just for *vfs_lwip*. Its awkward but less work than simultaneously removing *libc_lwip* or compiling LwIP to support both APIs and ensuring that *libc_lwip* still works.

chelmuth, 2017-12-20:
Let's follow the original plan of replacing libc_lwip by vfs_lwip and libc_lxip by vfs_lxip (which is already there). I'm afraid if we hesitate to do this step we will again miss to integrate the work into the next release.

What about the other bullet points above? Could you address them?

ehmry, 2017-12-20:
- libc_lxip has to be removed - I would like to move noux to libc_lxip for the moment, noux may be the most complicated part of fully integrating socket_fs

- nic_dump and nic_router run scripts have to be adapted - I have these working now on linux, nic_dump does not work for me on NOVA, I cannot test the ARM boards. This scripts I have a hard time understanding.

- http_srv_static test should be merged into http_srv (keeping configurability) - I will do
- http_srv_connect test tool can be removed - I will do

- tests adapted from lwip to libc API should be cleaned from references to lwip - I believe this is complete, with the exception of noux_net

- f050ce9 libc: internalize socket support as socket_fs - This is significant to noux, I will try and fix the low hanging problems above and then try to apply this patch again and look at noux

- f43fb55 lib/vfs/lxip: poll all open handles continously 6fff45e libc: default socket_fs path to '/socket' - perhaps this is not significant

chelmuth, 2017-12-21:
> 6fff45e libc: default socket_fs path to '/socket' - perhaps this is not significant

I'd like to have one pattern applied in all our scripts/scenarios. I'm fond of using */socket* as I don't see any device in sockets but if there's a majority for */dev/socket* I'd submit. Opinions?

ehmry, 2017-12-22:
I have a branch without libc_lwip or libc_lxip and changes to Noux at `issue2535_noux`.

ehmry, 2018-01-02:
I'm keeping all #2335 and #2535 changes on my `socket_fs` branch. Tests aren't going to stay updated if they change on staging.

chelmuth, 2018-01-03:
@ehmry it seems your topic branches for #2335 and #2535 are gone but there's no *socket_fs* branch. Did you forget to push the new branch?

ehmry, 2018-01-03:
It seems I did not push it.

commit 0e7d706b (2017-10-19): libc: internalize socket support as socket_fs

Remove libc_lxip, libc_lwip, and noux/net. Remove socket support from
libc plugin interface. Bind socket support directly to socket_fs.

Fix #2535
commit 4108e9cd (2017-10-20): libc: internalize socket support as socket_fs

Remove libc_lxip, libc_lwip, and noux/net. Remove socket support from
libc plugin interface. Bind socket support directly to socket_fs.

Fix #2535
commit 26d0158e (2019-01-31): Remove libc_lxip

The libc_lxip library is superceded by vfs_lwip.

Fix #2960
Fix #2535
commit 47c6377a (2019-02-12): Remove libc_lxip

The libc_lxip library is superceded by vfs_lwip.

Fix #2960
Fix #2535
closed 2019-02-12 by nfeske