genodelabs/genode #2555: fb_boot_drv: framebuffer pitch information missing

alex-ab, 2017-10-30:
On native hardware the pitch may be not the same as width * bytes per pixel, which results in scrambled video output. (kind of related to #2486)

alex-ab, 2017-10-30:
@jklmn: may you please give it at spin on your machines, if possible. I removed the 16 byte pad heuristic, because it should be covered now.

jklmnn, 2017-11-22:
Sorry for the late answer. It seems you accidentally misspelled my name so I didn't get notified. I will test it on occasion.

alex-ab, 2017-11-22:
@jklmnn: indeed, I misspelled your name, sorry for that.

jklmnn, 2017-11-23:
I tried the demo and it worked flawlessly.

What I noticed were differences with the framebuffer test. It seems that it had some shifts on the device. I experienced this with both versions, my padding and your pitch implementation.

Framebuffer test stripes in qemu (800x600):
[image]

Framebuffer test stripes on Lenovo Yoga 300 (1366x768):
[image]

Since the demo with nitpicker is working this could also be a problem with the framebuffer test.

chelmuth, 2017-11-23:
I'd expect the error to be in *fb_boot_drv*'s blitting code as the framebuffer test is completely pitch-agnostic (because `Framebuffer::Mode` has no pitch information).

chelmuth, 2017-11-23:
Sorry, I was too hasty. The strange striping pattern originates indeed from the pattern-drawing code of the framebuffer test, which the following patch for pitch enforcement in *fb_boot_drv* reveals also under Qemu.
```diff
diff --git a/repos/os/src/drivers/framebuffer/boot/framebuffer.cc b/repos/os/src/drivers/framebuffer/boot/framebuffer.cc
--- a/repos/os/src/drivers/framebuffer/boot/framebuffer.cc
+++ b/repos/os/src/drivers/framebuffer/boot/framebuffer.cc
@@ -33,6 +33,7 @@ Session_component::Session_component(Genode::Env &env,
 		fb.attribute("bpp").value(&_core_fb.bpp);
 		fb.attribute("pitch").value(&_core_fb.pitch);
 		fb_boot_type = fb.attribute_value("type", 0U);
+		_core_fb.width -= 3;
 	} catch (...) {
 		Genode::error("No boot framebuffer information available.");
 		throw Genode::Service_denied();
```
I added a patch from one of my development branches which draws a one-pixel-wide frame around the framebuffer (b0f68b407f) to reduce the confusion.

jklmnn, 2017-11-23:
I tried this on the Yoga and the frame was correctly placed. But the stripes are still shifted and  can't say if this is intended at this point.

chelmuth, 2017-11-23:
Yep, that's intended as I did not fix the stripe algorithm which does not use the notion of rows and columns but a fairly simple calculation based on `Mode::width()`.

commit 1ea0d94b (2017-11-09): fb_boot_drv: evaluate fb pitch

Fixes #2555
commit 96723297 (2017-11-24): fb_boot_drv: evaluate fb pitch

Fixes #2555
closed 2017-11-30 by chelmuth