genodelabs/genode #2563: core: memory gets freed up in pd_session_component and ram_dataspace_factory which is used afterwards

alex-ab, 2017-11-05:
pd_session_component.h:
```
_constrained_md_ram_alloc(*this, *this, *this),                     
_sliced_heap(_constrained_md_ram_alloc, local_rm),                  
_signal_broker(_sliced_heap, ep, ep),                               
_ram_ds_factory(ep, phys_alloc, phys_range, local_rm, _sliced_heap),
```
ram_dataspace_factory.h:
```
Tslab<Dataspace_component, SLAB_BLOCK_SIZE> _ds_slab;   
```

In the destructor of Ram_dataspace_factory memory get freed up which is referenced by _ds_slab and _sliced_heap, which later on gets destructed.

#2562 gets blocked by this issue.

alex-ab, 2017-11-05:
The commit f22f5adbe1521e763eb08ce77f25b94c1b67e98a, in my opinion, don't qualify as fix, but rather showcase the dilemma.  With the commit the _revoke_ram_ds gets deferred until the point it is safe. Unfortunately for this to work, somewhere (enough!?) memory is required from somewhere. The other (theoretically) issue is that actual already during `ds->detach_from_rm_sessions()` the mapping in core could vanish (which in the current implementation does not happen, see Core_region_map::detach).

alex-ab, 2017-11-09:
Please have a look for 9bb0f571c892d074c4f270c539f9e8942b758e36. It helps for the _ds_slab from ram_dataspace_factory.h which already solves #2562. For the _sliced_heap one would need to have however another 'frontend' (to use just a Allocator interface) instead of Ram_allocator (requiring dataspaces and region manager). Opinions ?

alex-ab, 2017-11-09:
Actually, no, not a new 'frontend' for the heap - since it just consists of dataspace handling ...

nfeske, 2017-11-14:
Merged to staging.

commit dad121ed (2017-11-14): core: use core mem allocator in Pd_session

Fixes #2563
closed 2017-11-30 by chelmuth