genodelabs/genode #2569: Vfs::Dir_file_system allocates handles during read and sync

ehmry, 2017-11-07:
Dir_file_system, the VFS plugin that makes composite logical directories over other plugins, is allocating additional VFS handles during read and sync operations. Issue #2553 makes this problem noticeable when the VFS server fails to complete a read due to insufficient RAM for a temporary, internal directory handle. My solution is to open and cache handles on all member directories during `opendir` at Dir_file_system before a read a sync is issued, which allows the VFS server to catch an OOM error and pass it on to the client before the directory is read.

ehmry, 2017-11-07:
Fixed on my `issue2569_dir_file_system` branch.

ehmry, 2017-11-12:
@chelmuth I have a fix for the empty directory issue, 976e56f6d4a42c68ed0c8132b39706282cbe49dd. I wasn't handling the root directory the same as other directories and didn't notice wasn't testing readdir over the root directory.

nfeske, 2017-11-17:
Yesterday I rebased the sculpt scenario (https://github.com/nfeske/genode/commits/sculpt) on the current staging branch, which includes @ehmry's fixes of this issue. Unfortunately, the problem with invisible directory entries is still there but in a different (very easy to reproduce) way. When starting the sculpt scenario, the initial noux instance has several directory entries mounted at '/', in particular 'config/' and 'report/'. But 'ls' only shows 'lib/' and 'share/' as provided by the tar archives of the used noux packages. Strangely, other content of these tar archives is invisible too (e.g., 'bln/'). The issue can be reproduced in Qemu.

ehmry, 2017-11-17:
I've seen something like this as well, I believe `<dir/>` nodes are disappearing.

ehmry, 2017-11-18:
Fixed at f3281d0593665607b50f9d890f395085fdb0daa0. It was a simple problem of `opendir("")` not being treated the same as `opendir("/")`, so I went over Dir_file_system and removed any cases I found where a zero-length path would propagate.

nfeske, 2017-11-18:
Thank you for the quick response. However, your fixup does not solve the problem in sculpt. Admittedly, I have trouble following the changes because so many conditions are tweaked.

Together with @ssumpf we had a close look at the dir-file-system on Friday. I sense that you somehow changed the semantics of "/" and "" of my original implementation, maybe not by intention. @ssumpf started to revise the code to make it more clear. I.e., we came to the realization to abandon the "" argument altogether and instead equip the root directory instance of the dir-file-system with an attribute (using a separate constructor for creating the root directory). After all, the "" argument was solely needed to distinguish the root from non-root dir-file systems. This was a silly idea of mine which led to much confusion. I hope that this will make the code easier to follow. The remaining rules are:

* If it is the root directory, the dir-file-system inquiries the dir entries of all mounted file systems, passing "/" as argument.
* If presented with a "/", the dir-file-system returns its own name as a single dir entry.
* If presented with a path of one or more elements, the dir file system matches the first element against its own name. If different, it returns no dir entries. If they match, it strips its name from the front of the path and inquiries the dir entries of all mounted file systems with the remaining path,

@ssumpf started to review and change the code according to this.

ehmry, 2017-11-18:
Ok, good. I didn't realize that "/" and "" meant different things so that was not taken to account in the last few patches of mine.

ssumpf, 2017-11-21:
@ehmry: Could you please revert  ade9e26 and check if 048418a works for you?

ehmry, 2017-11-22:
@ssumpf Yes, works for me, thank you. I will make sure to make the same few fixes at world before the next release.

ssumpf, 2017-11-22:
@ehmry: Cool! Please base your fixes on ec01d34 - but I guess you have noticed this already.

commit dcfe8344 (2017-11-10): Restrict VFS Dir_filesystem allocations to 'opendir' only

Allocations made at the client-side of the VFS must be contained within
functions that may return errors indicating resource exhaustion.

Fix #2569
commit b2b9ed5b (2017-11-15): Restrict VFS Dir_filesystem allocations to 'opendir' only

Allocations made at the client-side of the VFS must be contained within
functions that may return errors indicating resource exhaustion.

Fix #2569
commit 972ef7bd (2017-11-16): Restrict VFS Dir_filesystem allocations to 'opendir' only

Allocations made at the client-side of the VFS must be contained within
functions that may return errors indicating resource exhaustion.

Fix #2569
commit ec01d348 (2017-11-21): vfs: handle root directory explicitly

This makes '/' and the actual root of VFS distinguishable. A VFS root
may contain one ore more '/' entries for each file system. 'opendir' for
the VFS root opens all file systems via 'open_composite_dir', while
'opendir' for '/' only returns a VFS handle.

Fixes #2569
closed 2017-11-30 by chelmuth