genodelabs/genode #2570: timeout api: attempt to insert alarm twice

nfeske, 2017-11-08:
While working with the sculpt scenario, I observed the following problem when interactively reconfiguring the input filter. The reconfiguration works several times but eventually, the input filter seems to die. In the log, I see suspicious messages:

```
Error: trying to insert the same alarm twice!
```

This points at an issue related to the key-repeat timeout handling. Whenever the input filter is reconfigured, most objects of the component - in particular the entire filter chain - are destructed and rebuilt. In this process, the timeouts as instantiated by the chargen filter should also vanish without any trace. However, when reconstructing the new filter chain, there seems to be a remains from the previous one referenced in an alarm scheduler(?). I guess that this problem culminates in the eventual hang of the component when the reconfiguration happens several times.

nfeske, 2017-11-08:
I managed to reproduce the problem with a modified input_filter.run test in commit https://github.com/nfeske/genode/commit/90bae257d51d5982950e98d20684d447c62c65aa

nfeske, 2017-11-08:
@m-stein Could you have a look?

m-stein, 2017-11-08:
Sure. I'm at it.

m-stein, 2017-11-08:
@nfeske: I think that's it:
cc63d8a input_filter: add missing virtual Source::~Source

nfeske, 2017-11-09:
@m-stein Thanks a lot for solving this mystery! Now I feel a bit bad about bothering you with such a silly bug of mine. ;-)

nfeske, 2017-11-09:
I can confirm that your fix solves the input-filter reconfiguration problem in the sculpt scenario. So now, one adjust all input-filtering parameters without any fear.

commit c7658f5f (2017-11-09): input_filter: add missing virtual Source::~Source

Previously the destructors of classes derived from Source were not
called when destructing a Source because Source missed the declaration
of a virtual destructor. This caused at least problems when
reconstructing a Chargen_source with its Chargen_repeater that includes
a One_shot_timeout. In this case, the members of the Alarm behind the
timeout were not reset and the next time beeing used the Alarm appeared
to be still active which wasn't true.

Fixes #2570
commit 411601eb (2017-11-24): input_filter: add missing virtual Source::~Source

Previously the destructors of classes derived from Source were not
called when destructing a Source because Source missed the declaration
of a virtual destructor. This caused at least problems when
reconstructing a Chargen_source with its Chargen_repeater that includes
a One_shot_timeout. In this case, the members of the Alarm behind the
timeout were not reset and the next time beeing used the Alarm appeared
to be still active which wasn't true.

Fixes #2570
closed 2017-11-30 by chelmuth