genodelabs/genode #2575: wifi_drv: libc suspend() called from non-user context

BorisMulder-CSL, 2017-11-13:
Hi people,
When I run the wifi_drv in either my own scenario or using make run/wifi, it crashes with this error. I tried building it on different machines with clean genode repos, on qemu and on hardware. Anyone knows what is going on? The configuration of wifi_drv is the same in my own scenario as it is in the run/wifi test scenario.

Full log of wifi_drv:

2017-11-13T16:21:44.175591+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Error: no plugin found for fcntl(2)
2017-11-13T16:21:44.175686+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Error: no plugin found for write(2)
2017-11-13T16:21:44.184171+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Reload wpa_supplicant configuration
2017-11-13T16:21:44.234427+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Warning: wait_queue_head_t is empty, wq: 0x1549a08 called from: 0x14861f4
2017-11-13T16:21:44.234538+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] INFO: NET: Registered protocol family 16
2017-11-13T16:21:44.234803+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Warning: wait_queue_head_t is empty, wq: 0x1549a08 called from: 0x1485475
2017-11-13T16:21:44.234951+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] INFO: NET: Registered protocol family 17
2017-11-13T16:21:44.235135+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Warning: wait_queue_head_t is empty, wq: 0x1549a08 called from: 0x1485475
2017-11-13T16:21:44.235372+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Warning: wait_queue_head_t is empty, wq: 0x1549a08 called from: 0x1487c92
2017-11-13T16:21:44.235551+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Warning: wait_queue_head_t is empty, wq: 0x1549a08 called from: 0x1484cb9
2017-11-13T16:21:44.236432+01:00 10.0.0.221  message repeated 5 times: [[init_1 -> init_2 -> wifi_drv] Warning: wait_queue_head_t is empty, wq: 0x1549a08 called from: 0x1484cb9]
2017-11-13T16:21:44.236712+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] INFO: Intel(R) Wireless WiFi driver for Linux
2017-11-13T16:21:44.236897+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] INFO: Copyright(c) 2003- 2015 Intel Corporation
2017-11-13T16:21:44.271590+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] ioremap: mapped phys 0xf1000000 (size 8192) to virt 0x16000
2017-11-13T16:21:44.286960+01:00 10.0.0.221  [init_1 -> rom_mapper] request for init_2 -> wifi_drv -> iwlwifi-7265D-16.ucode
2017-11-13T16:21:44.301869+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] dev_info: loaded firmware version 16.242414.0 op_mode iwlmvm
2017-11-13T16:21:44.302126+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] dev_info: Detected Intel(R) Dual Band Wireless AC 7265, REV=0x210
2017-11-13T16:21:44.305987+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] dev_info: L1 Enabled - LTR Disabled
2017-11-13T16:21:44.306857+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] dev_info: L1 Enabled - LTR Disabled
2017-11-13T16:21:44.313776+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Error: slab too large 131072reqested 102416 cached 0
2017-11-13T16:21:44.314286+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Error: libc suspend() called from non-user context (0x10fa506) - aborting
2017-11-13T16:21:44.314452+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Warning: 2 dangling allocations at allocator destruction time
2017-11-13T16:21:44.315116+01:00 10.0.0.221  [init_1 -> init_2 -> wifi_drv] Error: freeing non-empty slab block
2017-11-13T16:21:44.315948+01:00 10.0.0.221  message repeated 4 times: [[init_1 -> init_2 -> wifi_drv] Error: freeing non-empty slab block]

nfeske, 2017-11-14:
Can you please specify the branch / commit you are using? We recently added several wifi_drv-related fixes to staging, i.e., 7a359f95a33ae3eb55968e45f32e40572bb00770, but it is worth reviewing the current staging branch also for other wifi-related changes.

BorisMulder-CSL, 2017-11-14:
This was tested on the 17.08 release. I will look into the fixes on staging right now.

BorisMulder-CSL, 2017-11-14:
The above mentioned commit does indeed fix this now. Thanks for the response, I will try to keep up with staging better next time.

closed 2017-11-14 by BorisMulder-CSL