genodelabs/genode #2576: x86 platform driver does not fully constrain session capability quotas

ehmry, 2017-11-13:
To begin with, I have a problem with `intel_fb_drv` when booting from UEFI on my T420. The driver seems to do a lot of probing and then `platform_drv` blocks with a resource request to its parent for more capabilities, which blocks the boot process. The framebuffer driver is pushing the platform driver harder than usual, but the first concern is to protect the platform driver. I added some ram and cap quota guards in the platform session component and replaced the `Ram_session_guard` with a `Constrained_ram_allocator`. This moved the resource problem to the framebuffer driver, and seems to reduce the number of capabilities to be accounted directly to the platform driver.

A log of `intel_fb_drv` initializing on my T420 after apply the patch: http://sprunge.us/Gahe

ehmry, 2017-11-13:
Patch at eaf0868a5f00a923028daec1697c62e3e89e4baa.

nfeske, 2017-11-14:
Looks good to me. @alex-ab Could you also take a look?

alex-ab, 2017-11-14:
I'm fine with it, thx @ehmry

nfeske, 2017-11-14:
Merged to staging.

commit 252a90e7 (2017-11-14): platform_drv: account session capabilities with a Cap_quota_guard

The platform driver is a critical component and must not allow sessions
to deplete its own resource quotas.

Fix #2576
commit 79d029ab (2017-11-16): platform_drv: account session capabilities with a Cap_quota_guard

The platform driver is a critical component and must not allow sessions
to deplete its own resource quotas.

Fix #2576
commit de5c0603 (2017-11-30): platform_drv: account session capabilities with a Cap_quota_guard

The platform driver is a critical component and must not allow sessions
to deplete its own resource quotas.

Fix #2576
closed 2017-11-30 by chelmuth