genodelabs/genode #2579: timeout: reduce number of Time_source::curr_time calls

m-stein, 2017-11-15:


m-stein, 2017-11-15:
Currently, the number of calls to the potentially expensive Time_source::curr_time in the timeout framework is greater than it must be. At least the calls in Alarm_timeout_scheduler::_schedule_one_shot and Alarm_timeout_scheduler::handle_timeout are not necessary.

<b>Alarm_timeout_scheduler::_schedule_one_shot</b>
Here, we call curr_time to feed Alarm_scheduler::handle. We must call the latter before setting up a timeout because of the following problem.
+++
Assume we would not call Alarm_scheduler::handle before installing a timeout.
A user of the timeout framework may say: "schedule absolute timeout 0xffffffff" and expects the timeout to trigger far in the future. The Timeout framework must translate this relative to an absolute value to be able to install it at the Alarm framework. Let's say the absolute time for the timeout renders as 0x300 while the now time of the Timeout framework is 0x301. So, for the Timeout framework, the timeout is obviously in the next time counter period. However, in the Alarm framework, the now time may still be 0x299. This is not much off the real value (as it often gets updated) but it is still outdated enough that the Alarm framework must assume that the timeout is in the current time period. Thus, it would trigger much earlier than expected by the user.
+++
This problem can be fixed by using only relative timeout values outside the Alarm framework. This would slightly change the Alarm_scheduler interface but AFAIK this interface isn't used anyway outside the Timeout framework. I would nonetheless delay this solution until after the next release as it would influence nearly all Genode scenarios.

alex-ab, 2017-11-16:
Please have a look on my [issue_2579](https://github.com/alex-ab/genode/commits/issue_2579) branch and consider the commits as stop gap solution until this described issue gets addressed. The commits fixes PIT timer instabilities for me on native hardware and in qemu for okl4, pistachio, sel4 32/64bit.

m-stein, 2017-11-17:
@nfeske @alex-ab I uploaded a slightly adapted commit series:
96b22a6 timer: clamp one-shot timeouts to avoid overflow
1e70548 nova: limit timeout rate in nova_timer_drv
e741755 timer.run: provide more information if test fails
fa4235c timer: read PIT timer solely after interrupt
32eb9c0 timer framework: use Time_source::curr_time()

m-stein, 2017-11-17:
As a reminder for us in this issue - these commits are quickfixes that should be replaced later by a cleaner solution:
96b22a6 timer: clamp one-shot timeouts to avoid overflow
fa4235c timer: read PIT timer solely after interrupt
1e70548 nova: limit timeout rate in nova_timer_drv

The current plan consist of two parts:
* Merge the Alarm framework completely into the Timeout framework
* For all timers limit the rate in which the timer hardware throws interrupts (minimum length for timeouts that get installed at the timer hardware)

m-stein, 2017-11-20:
@alex-ab Could you please give those a short review:
975f9d1 timer nova: more clear max-timeout calculation
548cd8c fixup "timer: clamp one-shot timeouts to avoid overflow"

m-stein, 2017-11-20:
@chelmuth @alex-ab Adapted the commits to Alex' suggestions, so please merge:
f674224 fixup "nova: limit timeout rate in nova_timer_drv"
548cd8c fixup "timer: clamp one-shot timeouts to avoid overflow"

alex-ab, 2017-11-20:
@chelmuth: I'm fine with the changes.

chelmuth, 2017-11-20:
Nice, will merge them to staging if the following warning is fixed:
```
.../repos/os/src/drivers/timer/include/session_component.h:72:15: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     us = ~0UL >> 1;
```

skalk, 2019-02-12:
What is the progress of this issue? Aren't above commits already applied to upstream branches?

m-stein, 2019-02-12:
It's hard to tell. Some of the circumstances that motivated commits like 9d0f58d1052aaf949b38c28cc89fe267c5b77a84 or 759d39d2757f301185e7b193f336b256ea68f5c4 are no longer given because the timing stack significantly changed since then. As the issue is additionally older than the Greifensteine, I think we can close it and proceed.

closed 2019-02-12 by chelmuth