genodelabs/genode #2587: ahci_drv: scan_ports off-by-one corner case

MartijnVerschoor, 2017-11-26:
On an HP480 G3, the ahci port scan fails. In the log below note that the 'number of ports' is 2 while the 'pi' (ports implemented) map is 5 (b'101'). According to this map there are 2 ports: #0 and #2. Port #0 has an unknown device signature (well; 0xffffffff), and port #2 is the 'ATA' port we need to use.

[ahci_drv] --- Starting AHCI driver ---
[ahci_drv] AHCI found (vendor: 32902 device: 40195 class: 67073)
[ahci_drv] version: major=0x1 minor=0x301
[ahci_drv] command slots: 32
[ahci_drv] native command queuing: yes
[ahci_drv] 64-bit support: yes
[ahci_drv] number of ports: 2 pi: 0x5
[ahci_drv] #0: off (unknown device signature)

Currently the `ahci_drv` function '[scan_ports](https://github.com/genodelabs/genode/blob/e39e8c101dbb76fc96bc83e6c80eee77a1bd7c81/repos/os/src/drivers/ahci/ahci.cc#L123)()' doesn't inspect port #2 because it lies one behind the maximum port index that is inspected in the pi map. Therefore it fails to find any suitable port. If we extend the port scan with one iteration, port #2 is found and can be used.

I tried to pin what would be the correct behavior. 

The AHCI [specification](https://www.intel.com/content/www/us/en/io/serial-ata/serial-ata-ahci-spec-rev1-3-1.html ) states: 

- (on page 17) "Number of Ports (NP): 0’s based value indicating the maximum number of ports supported by the HBA silicon. A maximum of 32 ports can be supported. A value of ‘0h’, indicating one port, is the minimum requirement. Note that the number of ports indicated in this field may be more than the number of ports indicated in the PI register."

- (on page 19) "The maximum number of bits set to ‘1’ shall not exceed CAP.NP + 1.".

In our case CAP.NP = 1, thus the maximum number of ports is 2. This corresponds to the log. To me it seems that the `ahci_drv` correctly implements the specification. Of course, that doesn't solve our corner case.

The linux [ahci driver](https://github.com/torvalds/linux/blob/844056fd74ebdd826bd23a7d989597e15f478acb/drivers/ata/ahci.c#L1694) mentions:

	/* CAP.NP sometimes indicate the index of the last enabled
	 * port, at other times, that of the last possible port, so
	 * determining the maximum port number requires looking at
	 * both CAP.NP and port_map.
	 */

This comment indicates that the spec is either ambiguous or is not always followed. Assuming that with CAP.NP they mean the 1's based value (CAP.NP register value + 1), then this would solve our corner case.

ssumpf, 2017-11-26:
@MartijnVerschoor : I see two ways here, either start to add quirks (which will happen with non-standard conform hardware in the future anyways), or add your suggestion -  which I would prefer in this case. Thanks for the very thoughtful inspection! So,  please push you fix this issue. Thx.

chelmuth, 2017-11-26:
@MartijnVerschoor Your report reminds me of issue #2522 @ssumpf fixed in September with ff800af820e9ff5d1cd6cc148dda98e8947dc348. Are you using a Genode version that includes this commit? You're not using master for sure as your missing 73f1fa9a1f3626363a7a3499431a087a6f52b215.

Also, reading the spec snippets above it sounds fairly reasonable to me that port #2 should be detected according to the spec as PI=0b101 has not more than (CAP.NP=1)+1 bits set.

ssumpf, 2017-11-26:
@chelmuth: You're right #2522 should fix this issue, thanks for the reminder and for keeping the overview!

MartijnVerschoor, 2017-11-27:
@chelmuth, you're right, it's a duplicate of #2522. The issue popped up in a scenario that runs on 17.08 with a minimum of cherry-picks from master/staging. I should have inspected changes in ahci_drv before reporting the issue, but didn't under the assumption that I found something rare.

Inspecting the driver code and the ahci specification did give me some insight into a component that I'd normally take for granted.

The spec is a bit ambiguous regarding CAP.NP. At the very least it is clear that the number of bits set to 1 in the PI register cannot exceed CAP.NP + 1. Besides this, the spec hints that CAP.NP also determines the maximum index of available ports (e.g. in the example in section 3.1.4). This is correlates to the previous implementation and the quoted comment from the linux driver, but is probably not right.

@ssumpf, I reviewed ff800af820e9ff5d1cd6cc148dda98e8947dc348 and agree with your approach to not make any assumptions on the index of available ports, yet to assert that the number of ports does not exceed CAP.NP + 1. I'd say this is the most strict interpretation of the spec.

It solves the issue on the mentioned platform. 

Thanks to both of you for your quick response! I'll close the issue.

closed 2017-11-27 by MartijnVerschoor