genodelabs/genode #2588: intel_fb_drv: restarting it or its clients fails

BorisMulder-CSL, 2017-11-27:
When I have an init component that dynamically restarts its children, using it together with intel_fb_drv fails.

- If I put intel_fb_drv as a child of this init (and restart it together with its children), the driver stops, the screen turns into a noisy bit of pixels, and my system becomes extremely slow while spamming thousands of messages similar to the one below:
[ 0] DMAR:0xffffffff81036040 FRR:0 FR:0x7 BDF:0:2:0 FI:0xce1c9000
with many different FI values.
After a few minutes this stops, the screen goes back to where it was before the restart, but the driver does never announce the Framebuffer service and the clients (i.e. nitpicker) are blocked forever.

- If I put intel_fb_drv below this init in the component tree, do not restart it but restart only its client, nitpicker, it never acquires the Framebuffer session and the screen keeps hanging in the noisy state.

- If I use the vesa driver in either of these scenarios, it works fine.

chelmuth, 2017-11-27:
I had a quick look into intel_fb_drv and suspect the issue in https://github.com/genodelabs/genode/blob/master/repos/dde_linux/src/drivers/framebuffer/intel/include/component.h#L192 This code of the root component returns a pointer to the `session` member on `_create_session()`. Unfortunately, the root component does not override `_destroy_session()` which calls `destroy(..., session)` in its default implementation effectively calling the destructor of the `session` member. A subsequent session request then uses a destructed `Session_component`.

In short: Please try to override with
```c++
void _destroy_session(Session_component *session) override { }
```
to test my suspicion.

BorisMulder-CSL, 2017-11-28:
This change does fix the second problem. The first one is not fixed, though. I do have #2564, and it should not be a problem of platform_drv as the vesa driver does not have this issue.

alex-ab, 2017-12-01:
Restarting drivers is not a explicit supported feature nor tested. For simple drivers you may have just luck (vesa qualifies for this). To make the restarting of real/complex drivers more robust, we would need some more support at least by the platform driver and potentially partially by the kernel. Therefore your first item will not be fixed until we added explicit restart support for drivers. I would suggest to open a feature issue (for discussion and testing the targeted drivers) and close this issue, if the client part is sufficiently fixed.

jddtelnet, 2017-12-01:
I am way out of my league here, but could the reincarnation server from
Minix be ported to Genode? Or is it too closely tied to the Minix
microkernel to serve any purpose for Genode components...?

alex-ab, 2017-12-04:
I suppose it is tied to Minix. I expect that squashing Genode specifics into it, if it is not designed for various  back-ends in the first place, effectively comes down to a complete rewrite. In principle we have most in place (init can restart components, so also drivers). At the moment we need a feature to quieten/turn off the device when a driver forcefully vanishes instead of let continuing to run the device (and causing therefore load on the hardware and tons of warnings which effectively slows down the system.)

BorisMulder-CSL, 2018-01-30:
@chelmuth May I request your _destroy_session override to be merged with staging or master if appropriate?

chelmuth, 2018-02-05:
@BorisMulder-CSL thanks for the reminder. Merged as 8c94483d81d64e29a44911b19c9e593a7d19224c to staging.

BorisMulder-CSL, 2018-02-05:
Nice!

alex-ab, 2018-02-16:
I'm going to close this issue. The client closing part is fixed. The actual restarting feature of drivers will be considered in a feature issue over the year.

closed 2018-02-16 by alex-ab