genodelabs/genode #2590: nic_router: advanced timeout configuration

m-stein, 2017-11-28:


m-stein, 2017-11-28:
Fixed by:
ce29c95 nic_router: TCP keepalive interval to DHCP clients
f1f3ab3 nic_router: advanced timeout configuration
c54f7a6 nic_router: read seconds attributes generic

chelmuth, 2017-11-30:
I merged 564e6a688562fcc7cc510b8ba3f70c29a280a14b and d2bb51a5c949f4a8184de08371e1c1be015d74a4. As I'm not sure about all implications of ce29c95 I skipped it currently.

I have no experience with TCP KEEPALIVE and also never heard of the DHCP option. Do you know any background and esp. how it behaves in cascades of routers? Honestly, I hesitate because I'm afraid of unexpected effects on the variaty of vbox guest systems -  speak Windows 7/10 and flavors of Linux.

chelmuth, 2017-12-05:
I just noted that your *merge_to_staging* branch has some commits that may be worth merging. Unfortunately, it still contains the above mentioned TCP KEEPALIVE commit, which I'd like to skip from merge. Also, you did not reply to my concerns above.

chelmuth, 2017-12-11:
With the current changes the router works like a charm in my scenario. But I have a small change request: The router logs a warning for each configuration attribute that is not configured explicitly. IMO it is sufficient to have sane default values and remove the log message. What do you think?

m-stein, 2017-12-11:
Ok, what about changing the messages from warnings to verbose-only logs?

chelmuth, 2017-12-11:
I personally prefer less noise in the log.

m-stein, 2017-12-11:
So what do you think about my suggestion. This would show the messages only with verbose="yes".

chelmuth, 2017-12-11:
I'm uncertain about this direction. IMO a bad example is the verbose log of some of our dde_linux drivers which is almost unreadable noisy (even for the original authors after some months). Yes, we're discussing about 4 lines of log only in this specific example. But, the direction hints it is a good idea to write the detailed configuration of a component into the log, which I disagree with.

m-stein, 2017-12-11:
Ok, no problem for me. I'll remove it.

nfeske, 2017-12-11:
How about the following suggestion of a rule of thumb: Information may be logged if it depends on runtime behavior like the result of probing a device, the answer of a DHCP query, or state that depends on user input.  E.g., I like that the NIC router tells me the IP addresses because this information varies from run to run. So I use it in verbose mode. Static information like reflecting the configuration of a component, or the fact that the component was started at all, does not meet this criterion. In practice, such information makes the log more noisy but offers almost no insight.

The only scenario I can think of where such static information may be valuable would be the remote help of someone who sends us the log only. But in this case, the help-seeking person could also provide the config along with the log.

As so often, this is not clear cut, but maybe the rule of thumb above could be taken as a guideline?

m-stein, 2017-12-11:
Thanks @nfeske for this suggestion. There are quite some more messages that do not fit this criterion. For instance, the router currently prints domain configurations. This was thought to be useful when the router config (or the whole router start node) gets generated dynamically. Up to now, it served only for debugging the config parsing in early versions of the router. I planned to rework the logging of the NIC router in the near future anyway as there are several groups of messages  like:
* Packet dump
* Routing decisions
* DHCP-client events
* DHCP-server events
* ARP events

... that are partially not complete and also all combined under one verbose flag. But first I wanted to implement reports to see if log subjects and report subjects maybe have something in common.

m-stein, 2017-12-11:
@chelmuth:
23d31d0 nic_router: do not warn when using default values

m-stein, 2017-12-11:
Btw: This is the first commit I wrote, tested, and uploaded using my Genode system. As I forgot the SSD with my Linux configs at home this morning, it took me some time to set up the environment but it works like a charm :-)

nfeske, 2017-12-11:
That's fantastic to hear!

chelmuth, 2017-12-21:
Fixed in master by 3cdcb528ff8206511034239ba8ac2b335778642a

closed 2017-12-21 by chelmuth