genodelabs/genode #2594: base-foc: fix thread pause/resume

m-stein, 2017-11-29:


chelmuth, 2017-12-06:
I just merged 9b2255f and want to make sure that everybody agrees with the documented rationale of the change (as I do). So, does anybody like to complain? @cproc? @ssumpf? @skalk?

m-stein, 2017-12-06:
Thanks @skalk for this hint:
6f6c7c0 fixup "base-foc: fix bug in Platform_thread::pause"

cproc, 2017-12-07:
Perhaps the Fiasco.OC kernel patch still applies for the current version: https://github.com/skalk/foc/commit/79c2653ed362578ba060dacf70e2826f7b77ebdb

m-stein, 2017-12-07:
@cproc Thanks a lot for this hint! I applied it manually to the current FOC version and it makes work both run/cpu_sampler_noux and run/thread again.

@skalk: Please consider merging to the FOC contrib:
https://github.com/m-stein/foc/commit/ecca85a88e77d4477f0f45dd928acd5d2b07425f

@chelmuth Please consider reverting:
e74daf5 fixup "base-foc: fix bug in Platform_thread::pause"
9b2255f base-foc: fix bug in Platform_thread::pause

skalk, 2017-12-08:
@all thanks for the investigation, I've re-applied the patch from ssumpf

chelmuth, 2017-12-11:
Any idea what `Warning: failed to acknowledge exception, l4_ipc_err=2` in the cpu_sampler_noux log means? I don't know if the message was there before the changes, though.

m-stein, 2017-12-11:
This warning came with https://github.com/m-stein/genode/commit/09fe1cb1cdf5cda3219eba1030ee58d907b8eeda in context of this issue. It means that the l4_ipc_send syscall in our Ipc_pager::acknowledge_exception() failed. This was the case previously when resuming a thread after the broken pause call (without m-stein/foc@ecca85a). As the error condition did not imply a log message it made debugging more time consuming and I thought we should log it in general. I don't know why it still occurs but I'll have a look at it.

commit ee873ce2 (2017-12-08): base-foc: fix thread pause/resume

Fix #2594
closed 2017-12-21 by chelmuth