genodelabs/genode #2595: openvpn; update to 17.11

MartijnVerschoor, 2017-11-30:
Recently I continued with the update of the openvpn component for more recent Genode versions (using with_libc) and lxip. The intermittent result is a working openvpn scenario that based on master 491be000ca2c07dbda8102a6c11104230cd91ad2 (just prior 17.11).

The udp/tap openvpn setup runs stable as far as I can tell. Over the course of two weeks I tested on NOVA x86_64 with real hardware and physical LAN. During this period we ran various iperf tests (averaging about 30 Mbit of throughput without optimizing). No failures are registered so far. Note that the version we tested was based on staging 01a328dc5f144fb890b4f8af9a74e869bca3ebda.

I would like to share the update and discuss our findings and modifications to get it to work. In particular, I introduced two workarounds that I believe should not be required; see further down below.

The work is uploaded to our github [fork](https://github.com/nlcsl/genode/commits/openvpn_master_just_prior_17.11). The first commit f01ccdb6be3d00fe3476bd44048343a84d753d74 is a reworked version of our first attempt to rebase openvpn to Genode 17.05 in June. We discussed this work on the [mailinglist](https://sourceforge.net/p/genode/mailman/message/35908946/) . At that time @chelmuth  contributed a speedy refactoring introducing an additional entrypoint e278fcad1e18ebefee02af70cde4c1876e4b5c47. This was a step in the right direction, but, as described on the mailinglist, there were some remaining problems. There were also other projects and summer holidays, so I paused the work.

About 6 weeks ago I was able to pick it up again. I noticed that the new entrypoint @chelmuth introduced was not forwarded to the ``Nic::Session_component``. In fact there is no option to pass a specific entrypoint in the constructor of the ``Nic::Session_component``. Instead the constructor retrieves the (rpc) entrypoint from the env argument. I added another constructor to fix this a4fba96c95b9a712dcf4fc58390b940765a94cf3 and refactored openvpn to forward the appropriate entrypoint 48d22d532296718c62ca5c7190ce8dd917d36ce7.

Next I ran into some troubles related to the socket_fs_plugin with which I'm less familiar. Specifically with the ``select`` function and the use of ``recvfrom``. The wait function in the event loop of openvpn relies on ``select`` to be notified about events on sockets. The configured wait function registers the file descriptors and the selected events it wants to be notified of. It appears that ``select`` works as expected, i.e. when a socket has data available to read, openvpn tries to read it using ``recvfrom``. I noticed that the call to ``recvfrom()`` repeatedly returns EAGAIN, indicating that there is nothing to read, eventough select indicates that there is. I traced the origin of EAGAIN to the ``read`` call embedded in ``read_sockaddr_in`` which is called from ``recvfrom()``. I noticed that the file descriptor used in the ``read`` call differs from the file descriptor that is registered at select. I'm not sure if its correlated, but under the hood in the filesystem session layer, the data and sockaddr are represented by different nodes. Would this pose a problem? I fiddled a bit to force the read in ``read_sockaddr_in`` and eventually got it working a5bb9c53b8e7687ed38de5e717f12d0fbe5b86b3. Of course this is not the proper solution.

Another issue we found is that openvpn no longer writes packets to the tun interface implemented in the Genode component (i.e. write received packets to the client over the Nic session interface). When data can be written to the tun interface, the openvpn loop registers an ready-to-receive event on the file descriptor associated with the tun interface. This event never triggers thus openvpn is never signaled that it can write to the tun interface. A mechanism to apply back pressure from the client was never implemented, but before Genode 17.02 openvpn could always write. I introduced this 9d0f17b53e3ff51d7ea83769e043378a9d5762ee workaround in openvpn to circumvent this. See also the text in the commit message.

chelmuth, 2017-12-13:
I had a closer look at 2aba8d68a5523d350dc4c5b624d0b8c046e5dd18 because it bothered me that it is still required to tweak is this regard after all the sweat to get the `select()` semantics right. For me it looks like the current implementation still suffers from the fact that the VFS was not available at the time of the initial port. Now, I would suggest to implement the file-to-nic-session mechanics as a VFS plugin, which automatically renders it fit for all the libc magic including select. This direction will not completely remove the OpenVPN tun backend implementation as the the VFS plugin should not be tainted by libc (or even Linux) specifics like ioctl defines. What do you think?

chelmuth, 2017-12-13:
Regarding 77c3b83366358dff05ab1faeacabc6ee02686315 I am not completely certain about the scenario that may trigger the flawed behavior. Do you see a chance to reproduce it in the netty example?

MartijnVerschoor, 2017-12-14:
@chelmuth, thanks for picking this up. Regarding your comment above on 77c3b83366358dff05ab1faeacabc6ee02686315, I have written a small test scenario that roughly models openvpn's UDP socket use 5434c047ea8cd509853126be0aca342ee8c5e48b. It reproduces the problem I encountered. I also added a test program for Linux for comparison.

The program opens a socket for UDP and sets the O_NONBLOCK flag on its file descriptor with fcntl.

You'll see that ``select`` returns when there is something to read from the socket (correct), while the subsequent call to``recvfrom`` returns EAGAIN (incorrect) . On Linux ``recvfrom`` reads the data as expected.

The test program sends a UDP packet to the configured server and waits until a UDP packet is returned. A simple UDP echo server will suffice to test the scenario. I used ncat (from nmap):

```
ncat -vvv -e /bin/cat -k -u -l 10000
```

MartijnVerschoor, 2017-12-15:
Hi Christian, 

> For me it looks like the current implementation still suffers from the fact that the VFS was not available at the time of the initial port. Now, I would suggest to implement the file-to-nic-session [...]

I agree. It makes perfect sense.

I would say the goal shouldn't be to get rid of the Genode tun implementation. In fact it's perfectly fine to have a Genode specific implementation. There is no way around it as OpenVPN's tun.c is full of OS specific code. It would not make sense to try to emulate one of the already supported OS.

The goal should be to support ``select`` and friends on libc file descriptors backed by the file-to-nic-session mechanism. This would be a more natural fit for OpenVPN and (if I'm correct) would eliminate the need to pass semaphores over the additional libc pipe. It should yield bidirectional back-pressure on the tun and a leaner implementation of the Genode tun implementation.

Right now the Genode OpenVPN component is setup to provide the tun/tap nic as a service. I'm not sure how this fits in the VFS framework. For our purposes, we identified that requesting the tun/tap nic as a client is desirable since it enables us to start, stop and restart the OpenVPN component within a compound Genode network architecture. Thus I would promote implementing the VFS tap (or nic) plugin simply as client.

nfeske, 2020-04-23:
I'm closing this issue as it has become stale. In the meanwhile, we decided to relocated OpenVPN to the Genode-world repository (#3737).

closed 2020-04-23 by nfeske