genodelabs/genode #2600: run: target-specific config schemata

m-stein, 2017-12-03:


m-stein, 2017-12-03:
This is first fully working version:
3a1044a nic_router: add config schema
84a0d07 run: target-specific config schemata

@nfeske I'm not so sure yet with the changes made to the build system that are the link between the run tool and the XSD files in the source directories in 84a0d07 . Could you please have a look at this?

nfeske, 2017-12-04:
It looks very elegant! I very much like your idea of tying the xsd-file with the target.mk.

m-stein, 2017-12-04:
Fixed some minor flaws of the initial version. So, @chelmuth please consider merging:
6fc37a6 nic_router: add config schema
2dfab0b run: target-specific config schemata

m-stein, 2017-12-05:
There's a little fixup that came up when using only depots in a run script:
260c40d fixup "run: target-specific config schemata"

m-stein, 2017-12-07:
I just found out that we can do this also with xmllint only, so, xmlstarlet is not required anymore:
903539d run: avoid dependency for xmlstarlet

m-stein, 2017-12-07:
... 903539d may be used as fixup for:
bff23f3 run: target-specific config schemata

m-stein, 2017-12-08:
Another fixup: d29c374 fixup "run: target-specific config schemata"

commit 0931adb0 (2017-12-05): nic_router: add config schema

Fix #2600
commit fc7999a6 (2017-12-21): nic_router: add config schema

Fix #2600
closed 2017-12-21 by chelmuth