genodelabs/genode #2608: tool_chain: clean deletes whole build directory

jklmnn, 2017-12-11:
When running `tool/tool_chain clean` the script deletes the complete contents of the build directory (without any warning). The expected behaviour would be to only delete
 - bootstrap
 - gmp
 - install
 - lib-install
 - mpc
 - mpfr
 - test
 - x86 (probably architecture in general)

Generally it probably would be better to create either a separate build directory or a subdirectory within `build` to avoid collisions with existing directories.

nfeske, 2017-12-11:
This is really an unfortunate series of events. Originally, the `tool_chain` script was free-standing, not tied to `GENODE_DIR`. Back then, we had no convention of hosting the regular build directories at `<genode-dir>/build/`. The `tool_chain` script would download and build the tool chain at the current working directory (cwd) from where it was called. E.g., when executing the script from within `/tmp`, the download and build directories would be created at `/tmp/build` and `/tmp/download`. This is still the case. What happened now is that you happened to execute `tool_chain` from the root of the Genode source tree.

I wholeheartedly agree with your suggestion. Since we introduced the convention to host `build` and `contrib` at the root of the Genode tree, it would make much sense to adapt `tool_chain` to follow this convention as well.

nfeske, 2017-12-11:
The tool chain downloads could go to `<genode-dir>/contrib/tool_chain-<version>` and the build directory could reside at `<genode-dir>/build/tool_chain-<version>`. Here, `<version>` would correspond to the Genode version to which the tool chain belongs.

cproc, 2019-04-05:
Fixed in 27c566c

commit 20bb6fa2 (2019-04-09): tool_chain: build in <genode-dir>/build/tool_chain-<version>

Fixes #2608
commit 48708060 (2019-05-03): tool_chain: build in <genode-dir>/build/tool_chain-<version>

Fixes #2608
commit ed246d15 (2019-05-06): tool_chain: build in <genode-dir>/build/tool_chain-<version>

Fixes #2608
closed 2019-05-06 by chelmuth