genodelabs/genode #2613: Examination of packet stream signaling

ehmry, 2017-12-13:
To start with a concrete problem:

The `Packet_descriptor_transmitter` and `Packet_descriptor_receiver` classes both advertise an API that will submit or receive packets immediately or after blocking for a signal from the remote side. During construction of the `Packet_stream_tx::Client` and `Packet_stream_rx::Client` classes there are two signal capabilities each that are submitted via RPC to the remote end. The capabilities are for the `Signal_receiver` that is used to block the local end until the remote end signals that a queue is no longer empty and another to block until a queue is no longer full.

If the application were to send capabilities for `Signal_handler`s to the remote end, the capabilities to the local `Signal_receivers` would no longer be attached to the remote, and the application will block indefinitely should a packet be submitted to a full queue or to received from an empty queue, where those signal handlers are replaced. The problem is that this is not mentioned in the documentation, and it does not allow for a packet stream to be processed both by asynchronous signal dispatching and by synchronous application code.

This could be avoided by adding an RPC method to both `Packet_stream_rx` and `Packet_stream_tx` 
interfaces like `wake_sigh` that would install an additional signal capability at the remote end to be signaled whenever the packet stream left a blocked state.

Or to avoid using an additional capability at packet streams, the `Signal_receiver`s in the `Packet_descriptor_transmitter` and `Packet_descriptor_receiver` classes could be consolidated and moved up to the `Packet_stream_source` and `Packet_stream_sink` classes so that a sink and a source use only one signal each to notify a non-blocking state. The methods to register signal receivers for blocking could be hidden from the outward API and only used during construction of packet stream client construction. Then an extra signal could be added back (bringing it back to two signals per stream session) that would be for signal dispatching, and this would be a part of the outward packet stream API.

Changing the way the packet streams work is a bit radical but I think it flows with the new distinction between application signals and I/O signals.

nfeske, 2017-12-19:
Your observations are spot-on. When I originally designed the packet-stream, the synchronous way of programming was still pretty much ingrained in my head. If I designed it now, it would look different:

* It would not contain built-in signal receivers. No method would block. Instead, when encountering a state state where it would block today, the method would throw an exception.

* It would not distinguish the different control-flow signals. Instead a single signal handler can easily check all conditions (as illustrated by the nic_loopback component).

* It would defer the submission of a signal until after a batch of packets is inserted into the packet stream. E.g., right now, the first packet inserted into an empty packet stream immediately wakes up the sink. Depending on the kernel's scheduler, this may result in an immediate context switch - or not. It would be better to make the mechanism less reliant on the kernel's scheduling policy. I designed the implicit submission of signals in `submit_packet` as a precaution against someone missing to explicitly trigger data-flow signals. The user of the packet stream does not need to care about signal submission. At the API level, I would nowadays use a lambda-style way of ensuring that a signal is submitted after (potentially many) packets are inserted into a stream. If you like, I can elaborate more about this idea in a follow-up comment.

* Since most packet streams are client-initiated streams of requests, I would possibly reconsider the decoupling of the packet stream from the client/server roles as it complicates the terminology a lot (client/server, rx/tx, source/sink). It might be a much more intuitive model to just speak of request streams that always go from the client to the server. The data may still flow in either direction, like file-system read and write - both are requested by the client but the data flows in both directions. The only session interface where this does not fit perfectly well is the NIC session where two packet streams are currently embedded. But maybe even the NIC session could be re-modeled as a stream of (rx and tx) requests both initiated by the client?

Now, the question is how to get from the current version to a better one? Breaking API compatibility with the packet stream at once is not an option because it would force us to immediately adapt too many components. I rather contemplated the idea to introduce a new "request_stream" API, then step-by-step changing the existing session interfaces to use the new mechanism, and finally abandon the old packet-stream API.

Of course, alternatively, we might slowly reshape the current packet stream into something new. E.g., as one step, we might remove the built-in signal receivers. Then, as a next step, we might consolidate the signal types. Then, as another step, we might introduce a new packet-submit method that allows the insertion of batches of packets (by a lamda function passed as argument). We just need to be very careful to take small steps and complete one step (including the adaptation of all affected components) before starting the next one.

ehmry, 2017-12-19:
I agree that this is not something that should be rushed into, and I get understand what you mean about deferring signaling. However in the present environment packets streams must support blocking, unless services move to fully synchronous or fully asynchronous interfaces. In the simple case, some `File_system` servers need to perform an asynchronous `Block` operation to complete synchronous directory traversal. With this in mind, it may be sensible to move all `File_system` operations into packets if there is going to be a new "request_stream".

To give some input on a new request_stream, my reasoning for bringing this up is that I have a (almost working) `File_system` server backed by a content-addressed chunk store. Each file is made of chunks and directories are storing in chunks. The first awkward situation was designing the packet stream for the chunk store. The client requests chunks by hash but does not know the size of the chunk, so to make it work I have the server as the source and the client the sink. That means the client initiates a request with `acknowledge_packet`, the sever responds by allocating space in the shared buffer and uses`submit_packet`, and finally the client informs the server that the buffer space can be release with `acknowledge_packet` again. I don't see anything wrong with this yet, so I would hope the next packet stream terminology could somehow be rephrased into simpler terms relative only to the source of shared buffer allocation.

nfeske, 2017-12-27:
> In the simple case, some File_system servers need to perform an asynchronous Block operation to complete synchronous directory traversal. With this in mind, it may be sensible to move all File_system operations into packets if there is going to be a new "request_stream".

I didn't want to go as far - at least for now. There will always be cases where we need to block on a packet stream. My intention is to remove the built-in `Signal_receiver` from the packet stream. So instead of letting `get_packet` block. The packet-stream user would use a loop like:

```
while (!packet_avail())
  wait_and_dispatch_one_io_signal();
```

So the blocking semantics are completely up to the user of the packet stream. This change would eventually allow us to remove the `Signal_receiver` from the public API.

Regarding the move of further file operations to an asynchronous mode of operation, we should discuss this individually. I think that it should be pursued if there is good potential for simplification or performance improvement.

> The client requests chunks by hash but does not know the size of the chunk, so to make it work I have the server as the source and the client the sink. 

So in this case, the requests originate from the client but the allocation happens not at request time but at the server? Does the same request stream also need to support client-side allocations? If so, this may become very difficult because the allocator needs to be shared between client and server. Nevertheless, thanks for this interesting requirement. I'll need some time pondering about it.

ehmry, 2018-01-18:
I've reverted my experimental session to the standard client-allocated, server processed packet stream. Allocating packets at the server seemed like a way to save memory and stuff more packets in the buffer, but it resulted in two round trips and extra packet tracking effort.

nfeske, 2018-01-22:
All the better, then! Thanks for reconsidering your interface in the light of my comment.

nfeske, 2020-04-23:
There is no further work planned right now. We will eventually remove the blocking API functions but I'll open a separate issue once the time has come.

closed 2020-04-23 by nfeske