genodelabs/genode #2614: nic_router: reports with useful information

m-stein, 2017-12-14:


m-stein, 2017-12-14:
I would much appreciate if you help me defining the term "useful information" and the granularity in which the reporting shall be configurable. Up to now, I have the following suggestions for the report subjects:
1) The routers IP-config at a domain (address, gateway, subnet mask (dns-server, ...?) )
2) RX bytes, TX bytes (per session | per domain | for the whole router ?)
3) Dropped packet count (for the whole router | per domain (source) | per session (source) )

m-stein, 2017-12-15:
This is a first try:
0b53501 nic_router: report some useful information

With this, one can let report TX bytes, RX bytes, IPv4 interface, and gateway. Each per domain.

m-stein, 2017-12-17:
A fixup:
88a1676 fixup "nic_router: report some useful information"

chelmuth, 2017-12-21:
Merged to master by edf1f9d84953e7ca5b5bdb9bf1c66c07d3d3772c.

closed 2017-12-21 by chelmuth