genodelabs/genode #2616: base-linux: undeclared __sigword and __sigmask

ValiValpas, 2017-12-14:
On archlinux with glibc 2.26, I get the following error because the glibc header file bits/sigset.h does not seem to be installed anymore (and because the definition of `__sigword` and `__sigmask` moved to another file).

> In file included from /home/johanness/repos/genode/repos/base-linux/src/lib/base/child_process.cc:21:0:
/home/johanness/repos/genode/repos/base-linux/src/lib/syscall/linux_syscalls.h: In constructor ‘Lx_sigset::Lx_sigset(int)’:
/home/johanness/repos/genode/repos/base-linux/src/lib/syscall/linux_syscalls.h:383:27: error: ‘__sigword’ was not declared in this scope
    _value[__sigword(signum)] = __sigmask(signum);
                           ^
/home/johanness/repos/genode/repos/base-linux/src/lib/syscall/linux_syscalls.h:383:48: error: ‘__sigmask’ was not declared in this scope
    _value[__sigword(signum)] = __sigmask(signum);
                                                ^
/home/johanness/repos/genode/repos/base-linux/src/lib/syscall/linux_syscalls.h: In member function ‘bool Lx_sigset::is_set(int)’:
/home/johanness/repos/genode/repos/base-linux/src/lib/syscall/linux_syscalls.h:387:34: error: ‘__sigword’ was not declared in this scope
    return _value[__sigword(signum)] && __sigmask(signum); }
                                  ^
/home/johanness/repos/genode/repos/base-linux/src/lib/syscall/linux_syscalls.h:387:56: error: ‘__sigmask’ was not declared in this scope
    return _value[__sigword(signum)] && __sigmask(signum); }

ValiValpas, 2017-12-14:
I use commit 30235e2 to fix this issue for me.

nfeske, 2017-12-14:
Thanks for investigating and for the patch.

Is there any reason why we couldn't always use the local implementation of these fairly simple functions if we name them differently (e.g., using `lx_` as prefix)? It would be nice to avoid the `ifndef`.

As a matter of style, the macros should better be converted to static inline functions.

ValiValpas, 2017-12-15:
Well, I'm not that much into those parts of linux/libc to give a sound answer here. However, I found the corresponding commit in the glibc:

https://sourceware.org/git/?p=glibc.git;a=commit;h=a992f506ff7162da2afa5a6151cc6f15949ceef4

The commit message indicates that sigmask no longer has a system-specific definition. I therefore believe a local implementation for this functions should be fine.

chelmuth, 2017-12-19:
I followed @nfeske's suggestion and implement the functions as inline methods in the `Lx_sigset` class (c77682efeb17935592be48bc6ef8735f24cc6150).

commit 633fda28 (2017-12-18): base-linux: implement lx_sigmask/lx_sigword explicitly

In newer glibc versions, the '__sigword' and '__sigmask' macros are only
available in sigsetops.h which is not installed. These macros are not
system-specific, hence we implement them explicitly ourselves.

Fixes #2616
closed 2017-12-21 by chelmuth