genodelabs/genode #2618: nic_router: unhandled `No_tcp_packet` exception

nfeske, 2017-12-17:
I just observed the NIC router aborting with the following message:

```
[runtime -> nic_router] Error: Uncaught exception of type 'Net::Tcp_packet::No_tcp_packet'
[runtime -> nic_router] Warning: abort called - thread: ep
[runtime] child "nic_router" exited with exit value 1
```

I am clueless but maybe this message rings a bell for @m-stein?

m-stein, 2017-12-18:
I do not know why you receive a broken TCP packet but that the router exits on such packets is a bug and fixed by this:
be6302c nic_router: handle all "No_X_packet" exceptions

m-stein, 2017-12-18:
I found some problems with the above commit. I'll update it. So you can skip it for now.

m-stein, 2017-12-18:
Now this should be fine:
71aeeff nic_router: no Arp_packet constructor when sending
b4d5cb9 nic_router: handle all "No_X_packet" exceptions

chelmuth, 2017-12-21:
Fixed in master by 1cae5ec8f6f3825afcc53b47e3b6990d8edc73a1 and d6d0bcd960540e7a113f33e703a03962568227f0.

closed 2017-12-21 by chelmuth