genodelabs/genode #2619: run: use XSD files from depots too

m-stein, 2017-12-17:


m-stein, 2017-12-17:
My suggestion to fix this:
b6de326 run: use XSD files from depots too
ef7d9fa prg.mk: copy xsd files instead of linking them
df1d48f nic_router.run: import nic_router from depot

nfeske, 2017-12-18:
I'd advice against pulling the nic_router from the depot in the nic_router.run script, assuming that this run script is regularly used to develop and debug the nic_router itself. Instead, it would be sensible to pull all other parts of the scenario from the depot, excluding only those components that most likely need instrumentation. There are already quite a few run scripts (e.g., menu_view.run, wm.run, depot_query.run) that follow this pattern.

m-stein, 2017-12-18:
Thanks for this hint! I only added it to have a fast validation of my fix.

m-stein, 2017-12-18:
@nfeske I've adapted the commits according to your suggestions:
22a51eb run: use XSD files from depots too
330db4e nic_router.run: import base and init from depot
751fcaf run: let import_from_depot accept list variables

commit 52febc12 (2017-12-19): run: use XSD files from depots too

Fix #2619
closed 2017-12-21 by chelmuth