genodelabs/genode #2620: drivers/x86: enable pci bridges and memory access pci config command bit

alex-ab, 2017-12-18:


chelmuth, 2017-12-18:
Sorry for the accidental assigment change.

cnuke, 2018-01-23:
0a6dbda is also needed on systems using coreboot where the GSI is 0xff but the device is MSI capable.

chelmuth, 2018-01-25:
It can be seen that you enable MEM/IO/DMA in the PCI command register for all drivers step-by-step. Wouldn't it make more sense to configure this PCI device *capability* in the platform_drv (like lx_kit does)? Or are there objections (like DMA should only be enabled by the driver at a certain stage)?

alex-ab, 2018-01-25:
In principle I agree, that at lest the MEM and IO PCI bits can be handled by the platform driver. The bus master DMA bit I would not switch on by default, after the experience with the audio device and the DMA IOMMU fault storm. After the discussion with you, we may switch on the MEM and IO bits, when the drivers ask for the infos of the PCI bars. As it sounds, we would need to change the structural organization in the ahci driver a bit. Therefore, I propose to open up a feature issue taking care about all this and also the drivers already in use.

chelmuth, 2018-01-25:
To clarify my stance: I suggest to enable the IO resp. MMIO PCI command-register bits when an I/O port or memory resource is actually requested. But as you noted offline, this could be moved into a separate issue sketching the idea and listing needed adaptions like for *ahci_drv*.

closed 2018-02-10 by alex-ab