genodelabs/genode #2624: netperf tests: test nic_router throughput

m-stein, 2017-12-19:


m-stein, 2017-12-19:
My suggestion to fix this issue:
4806146 autopilot.list: add netperf_*_router tests
ad6227d netperf tests: test nic_router throughput

m-stein, 2017-12-19:
From a quick test:
```
 nova_x86_64_lxip_stream        93.59 MBit/s
 nova_x86_64_lxip_stream_router 93.31 MBit/s
 nova_x86_64_lxip_maerts        94.06 MBit/s
 nova_x86_64_lxip_maerts_router 90.66 MBit/s
linux_x86_64_lxip_stream        92.15 MBit/s
linux_x86_64_lxip_stream_router 70.24 MBit/s
linux_x86_64_lxip_maerts        39.58 MBit/s
linux_x86_64_lxip_maerts_router 33.43 MBit/s
```

ssumpf, 2017-12-19:
@m-stein: Looks promising

m-stein, 2017-12-20:
I'd like to do Gb tests, can you give me a hint which machine to use?

commit 1528e398 (2018-01-17): autopilot.list: add netperf_*_router tests

Fix #2624
commit 47b11bf9 (2018-02-09): autopilot.list: add netperf_*_router tests

Fix #2624
closed 2018-02-09 by nfeske