genodelabs/genode #2627: uniform IP configuration reporting

ehmry, 2017-12-20:
I see a problem now with the multiple IP sacks, the IP address isn't reported the same way for each and this can cause problems in run scenarios. I suggest defining a simple procedure in a header in the os repo to log the configuration.

I know how to do this for LwIP, but I'm not sure how to get the IP address of lxip without binding a socket first.

ehmry, 2017-12-20:
I have a message string defined at 3b41e0838f032b26a8656392716bf4ebf62dfa3a, but I have no opinion on the format or the message.

chelmuth, 2017-12-21:
We had a related discussion about information formerly logged by the nic_router in context of #2614. I'm not decided yet that the IP configuration falls into the same category but I'm certain that I don't want a special header in *os*. If the IP info would be reported we could use a report_rom with verbose configuration in automated tests though.

closed 2017-12-21 by ehmry