genodelabs/genode #2632: Nic_connection: destructing causes page/segfault

BorisMulder-CSL, 2018-01-04:
Hi all, (and by the way a happy new year to everyone!)

I noticed that when creating and destroying a Nic_connection, this often crashes the program.
This can be reproduced with the program below. To me this gives the following log:

> [init -> test_nic_reconstruct] construct number 1
> [init -> test_nic_reconstruct] construct number 2
> no RM attachment (READ pf_addr=0x18018 pf_ip=0x7c569 from pager_object: pd='init -> test_nic_reconstruct' thread='ep') 
> page fault, pd='init -> test_nic_reconstruct' thread='ep' cpu=0 ip=0x7c569 address=0x18018 stack pointer=0xa04fef18 qualifiers=0x4 irUwp reason=1

The program:
```
#include <base/component.h>
#include <base/log.h>
#include <base/heap.h>
#include <nic_session/connection.h>
#include <nic/packet_allocator.h>
#include <util/reconstructible.h>

namespace Test {
	class Main;
	using namespace Genode;
}

class Test::Main {

	private:
		Env &_env;
		Heap _heap;
		Nic::Packet_allocator _tx_block_alloc;
		Constructible<Nic::Connection> _nic;

		enum { BUF_SIZE = Nic::Packet_allocator::DEFAULT_PACKET_SIZE * 128 };

	public:
		Main (Env &env) : _env(env),
		  _heap( env.ram(), env.rm() ),
		  _tx_block_alloc( &_heap )
		{
			for (long i = 0; i < 3; i++) {
				log("construct number ", i);
				_nic.construct(_env, &_tx_block_alloc, BUF_SIZE, BUF_SIZE);
			}
			log("destruct");
			_nic.destruct();
			log("done");
		}
};

void Component::construct(Genode::Env &env)
{
	env.exec_static_constructors();
	static Test::Main main(env);
}
```
Just for your ease, here is the corresponding run script:
```
set build_components {
	core init
	drivers/timer
	drivers/nic
	test/nic_reconstruct
}

source ${genode_dir}/repos/base/run/platform_drv.inc

append_platform_drv_build_components

build $build_components

create_boot_directory

set config {
<config verbose="yes" prio_levels="2">
	<parent-provides>
		<service name="ROM"/>
		<service name="LOG"/>
		<service name="RM"/>
		<service name="CPU"/>
		<service name="PD"/>
		<service name="IRQ"/>
		<service name="IO_MEM"/>
		<service name="IO_PORT"/>
	</parent-provides>
	<default caps="200"/>
	<default-route>
		<any-service> <parent/> <any-child/> </any-service>
	</default-route>
}

append_platform_drv_config

append config {
	<start name="timer" priority="0">
		<resource name="RAM" quantum="1M"/>
		<provides><service name="Timer"/></provides>
	</start>

	<start name="nic_drv" priority="0">
		<binary name="} [nic_drv_binary] {"/>
		<resource name="RAM" quantum="10M"/>
		<provides><service name="Nic"/></provides>
	</start>

	<start name="test_nic_reconstruct" priority="0">
		<resource name="RAM" quantum="10M"/>
	</start>
</config> }

install_config $config

append boot_modules {
	core ld.lib.so init
	timer
	} [nic_drv_binary] {
	test_nic_reconstruct
}

append_platform_drv_boot_modules

build_boot_image  $boot_modules

run_genode_until forever
```

BorisMulder-CSL, 2018-01-04:
After some digging I pinpointed the fault to somewhere inside the Genode::Packet_allocator::remove_range function.

alex-ab, 2018-02-13:
@BorisMulder-CSL: thanks for the testcase. I could reproduce it and the commit 9d91b264432d1b7188ee1b73ee1c977d1e495e1f fixes the issue for me.

nfeske, 2018-02-13:
Thanks @alex-ab. Merged to staging.

BorisMulder-CSL, 2018-02-15:
Thanks, this seems to fix it for me too!

closed 2018-02-15 by BorisMulder-CSL

nfeske, 2018-02-16:
@BorisMulder-CSL Your feedback is appreciated. Please don't mind to leave issues open - even if your original problem is solved. Since the commit message is tagged with `Fixes #2632`, the issue will be closed automatically once the patch enters the master branch.

BorisMulder-CSL, 2018-02-16:
Ok, didn't realize that. Shall I reopen?

nfeske, 2018-02-16:
No. The patch entered the master branch some minutes ago, anyway. ;-)

commit 9444a450 (2018-02-13): os: reset packet_allocator members

Fixes #2632
commit e05a7080 (2018-02-15): os: reset packet_allocator members

Fixes #2632