genodelabs/genode #2633: native service support for Noux children

ehmry, 2018-01-05:
The only parent services that children of Noux are allowed to access are *Timer* and *Log*. I propose that Noux support a configurable whitelist so that Noux applications may provide access to Genode services in an interactive POSIX environment.

ehmry, 2018-01-05:
Fixed at 6c4e038d7d8d3da67aa6b20ca446011ca7c4c6ae. Could use some documentation but I don't have a great example yet for a run scenario.

nfeske, 2018-01-05:
I am a puzzled about your suggestion. Noux applications are meant to be like Unix processes. They should use files. For bridging the gap between Genode services and files, we have the VFS. I cannot see the benefit in promoting Unix processes to interact with Genode services directly. In contrary, I see  mostly trouble when blurring the line between both worlds.

May you give a bit of background about the problem you are solving?

ehmry, 2018-01-05:
I have a utility for my content-addressed storage stuff that communicates with a backend daemon over HTTP. The daemon is written in Go and not by me. I am able to compile the same utility for Genode if I swap the backend interface-object for one that connects to my native Genode service instead. In this case using the native service from Noux is convenient because I can avoid using TCP or engineering a file-system interface at the server.

I do however appreciate that could cause a big mess if used unwisely.

nfeske, 2018-01-08:
Couldn't you realize your utility as `Libc::Component` instead? Why do you need Noux? If you want to use Noux as a user interface for your utility, may it be possible to just mount the config of (and reports generated by) your utility in the Noux VFS?

I am afraid to further proliferate the use of Noux beyond its current role. In the mid term, I actually want to get rid of it. ;-)

ehmry, 2018-01-08:
I will wait until I can give you a demo next month :-)

closed 2018-01-08 by ehmry

nfeske, 2018-01-09:
Hehe, can't wait to meet you! :-)