genodelabs/genode #2634: init: inconsistent ROM routing for binaries and libraries

ehmry, 2018-01-05:
When init routes a child ROM request for a library the name of the library module is prefixed with the name of the child. When init routes a ROM request for the component binary, this request is not prefixed. I argue that while the request is solicited by init, it should be prefixed as if the child was active and made the request by itself. The reasoning is that if there are policies that route libraries using a label prefix, it would follow that binaries should be follow the same route.

As an example, consider adding a start element to a dynamic init instance without the child binary or libraries within the default route of init. Also, the binary or libraries may not be present on the system at the time the child was configured, and the routing label may be used to resolve and fetch them remotely. Of course it works to simply use *init* as the binary in this case and by convention wrap any non-*init* subystem in *init*. However, this makes resource management more complicated because the single-child init would probably be static and that would make resource upgrades difficult.

Experimentally I have found this possible on Linux, but with NOVA I had problems. I suspect that this may require support for client-side asynchronous session requests if init is going to route these requests through its own children.

nfeske, 2018-01-08:
I made this distinction on purpose. The binary and dynamic linker are what I call "environment ROM sessions" of the child. In contrast to sessions initiated by the child, those sessions are created by the parent and also invoked by the parent. If we would prefix the environment ROM sessions with the child's name, we could no longer express a policy where those sessions are routed differently from child-initiated sessions. This ability is important though. E.g., when starting a child that fetches its ROMs from a `fs_rom` server running as a sibling, we want to route all child-initiated ROM requests to `fs_rom`. But we don't want to do that for the environment sessions. Otherwise, the parent (init) would suddenly depend on `fs_rom`. The environment sessions can be routed using the `unscoped_label`.

That said, I agree that it is inconvenient to need two routes instead of one (like I did in depot_query.run): https://github.com/genodelabs/genode/blob/master/repos/gems/run/depot_query.run

So simplify configurations like this, I am considering to add a new attribute `label_last`, which matches the last element of the label (i.e., the ROM-module name for ROM sessions). A route with `label_last="ld.lib.so"` would match both a child-initiated ROM-session-request and the environment ROM session for the dynamic linker.

ehmry, 2018-01-11:
I'm inclined to agree with you now, I was imagining scenarios with packages of data bundled with software components. A little experimentation shows this to be impractical and better solved using some sort of scenario metadata to tie data to software.

I would welcome a `label_last` feature though, I do find myself making multiple routing rules to handle ROMs like `ld.lib.so`.

closed 2018-01-11 by ehmry