genodelabs/genode #2635: vfs plugin: serialize file system calls

ssumpf, 2018-01-08:
make calls into a file-system implementation thread safe.

ssumpf, 2018-01-08:
@chelmuth: c8bfc1d08a2ece8d8ad26a028a89770e64846076 is the rebased version of  40554f2e434cf7180c2fec023d79d31db99a16e9

nfeske, 2018-01-09:
Thank you for addressing this important issue!

I wonder, would it be too hard to avoid the macro - as a matter of adhering to our C++ programming style? The lock should ideally be explicitly embedded in a class (`Vfs_plugin`?) instead being created out of thin air.

Granted, this code is part of the libc, which does not follow our style anyway. So I am not sure if I should be so picky here. My reasoning is that our style evidently eases the maintenance of the code. So the more consequent we apply it (i.e., by staying away from macros), the easier our future life will become. ;-)

chelmuth, 2018-01-09:
I'll look into options for cleanup of the patch as I suggested this intermediate shortcut.

cproc, 2018-05-30:
Another deadlock in Arora. This time two threads are involved:

```
thread 1:

#0  pseudo_end () at .../repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
#1  0x00007f99e05ab511 in lx_futex (val=0, op=0, uaddr=<optimized out>) at .../repos/base-linux/src/lib/syscall/linux_syscalls.h:350
#2  thread_stop_myself () at .../repos/base-linux/src/include/base/internal/lock_helper.h:63
#3  Genode::Cancelable_lock::lock (this=this@entry=0x13e52a0 <select_cb_list>) at .../repos/base/src/lib/base/lock.cc:123
#4  0x0000000001364134 in Genode::Lock::lock (this=0x13e52a0 <select_cb_list>) at .../repos/base/include/base/lock.h:33
#5  Genode::Lock_guard<Genode::Lock>::Lock_guard (lock=..., this=<synthetic pointer>) at .../repos/base/include/base/lock_guard.h:40
#6  Libc::Select_cb_list::Guard::Guard (list=..., this=<synthetic pointer>) at .../repos/libports/src/lib/libc/select.cc:74
#7  Libc::Select_cb_list::for_each<select_notify()::<lambda(Libc::Select_cb&)> > (this=0x13e52a0 <select_cb_list>, func=...) at .../repos/libports/src/lib/libc/select.cc:105
#8  select_notify () at .../repos/libports/src/lib/libc/select.cc:197
#9  0x000000000137a403 in Libc::Io_response_handler::handle_io_response (this=<optimized out>) at .../repos/libports/src/lib/libc/task.cc:395
#10 0x000000000141c28c in Vfs::Fs_file_system::Fs_vfs_handle::_complete_read (this=0x7f99e84fed60, dst=<optimized out>, count=<optimized out>, out_count=@0x7f99e8eff458: 6) at .../repos/os/src/lib/vfs/fs_file_system.h:147
#11 0x0000000001415808 in Vfs::Fs_file_system::complete_read (this=0x7f99e84e0800, vfs_handle=0x7f99e84fed60, dst=0x7f99e8eff710 "udp/1\n", count=511, out_count=@0x7f99e8eff458: 6) at .../repos/os/src/lib/vfs/fs_file_system.h:1009
#12 0x000000000138649b in Libc::Vfs_plugin::Check::suspend (this=0x7f99e8eff460) at .../repos/libports/src/lib/libc/vfs_plugin.cc:528
#13 0x0000000001383162 in Libc::Pthreads::suspend_myself (timeout_ms=<optimized out>, check=..., this=0x13e5cc8 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+2504>) at .../repos/libports/src/lib/libc/task.cc:367
#14 Libc::Kernel::suspend (timeout_ms=<optimized out>, check=..., this=0x13e5300 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>) at .../repos/libports/src/lib/libc/task.cc:735
#15 Libc::suspend (s=..., timeout_ms=timeout_ms@entry=0) at .../repos/libports/src/lib/libc/task.cc:882
#16 0x0000000001389d1b in Libc::Vfs_plugin::read (this=<optimized out>, fd=<optimized out>, buf=0x7f99e8eff710, count=511) at .../repos/libports/src/lib/libc/vfs_plugin.cc:539
#17 0x000000000136ecff in Socket_fs::Context::_read_socket_path (this=0x7f99e46ea690) at .../repos/libports/src/lib/libc/socket_fs_plugin.cc:102
#18 Socket_fs::Context::Context (handle_fd=88, proto=Socket_fs::Context::UDP, this=0x7f99e46ea690) at .../repos/libports/src/lib/libc/socket_fs_plugin.cc:118
#19 socket_fs_socket (domain=domain@entry=28, type=type@entry=2, protocol=protocol@entry=0) at .../repos/libports/src/lib/libc/socket_fs_plugin.cc:824
#20 0x0000000001371777 in _socket (domain=domain@entry=28, type=type@entry=2, protocol=protocol@entry=0) at .../repos/libports/src/lib/libc/socket_operations.cc:252
#21 0x00000000012fb659 in addrconfig (pai=<synthetic pointer>) at .../contrib/libc-c9c17345f57cb2f827c271806df456d829e32ce3/src/lib/libc/lib/libc/net/getaddrinfo.c:1545
#22 getaddrinfo (hostname=<optimized out>, servname=0x0, hints=0x7f99e8effbb0, res=0x7f99e8effb88) at .../contrib/libc-c9c17345f57cb2f827c271806df456d829e32ce3/src/lib/libc/lib/libc/net/getaddrinfo.c:523
#23 0x0000000005651a22 in QHostInfoAgent::fromName (hostName=...) at .../contrib/qt5-1544f4490148dea1ca3c4d98c9504a4597be3e51/src/lib/qt5/qt5/qtbase/src/network/kernel/qhostinfo_unix.cpp:215
#24 0x00000000056509bc in QHostInfoRunnable::run (this=<optimized out>) at .../contrib/qt5-1544f4490148dea1ca3c4d98c9504a4597be3e51/src/lib/qt5/qt5/qtbase/src/network/kernel/qhostinfo.cpp:516
...
```

In frame `#12`, the vfs lock is acquired, then in frame `#10` `handle_io_response()` is called to notify anyone who might be suspending because of an empty packet allocator or a full submit queue. In frame `#8`, 'select_notify()' is called and in the end the thread waits for the lock of the select callback list.

```
thread 2:

#0  pseudo_end () at .../repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
#1  0x00007f99e05ab511 in lx_futex (val=0, op=0, uaddr=<optimized out>) at .../repos/base-linux/src/lib/syscall/linux_syscalls.h:350
#2  thread_stop_myself () at .../repos/base-linux/src/include/base/internal/lock_helper.h:63
#3  Genode::Cancelable_lock::lock (this=this@entry=0x13e8300 <vfs_lock()::_vfs_lock>) at .../repos/base/src/lib/base/lock.cc:123
#4  0x0000000001389a33 in Genode::Lock::lock (this=<optimized out>) at .../repos/base/include/base/lock.h:33
#5  Genode::Lock_guard<Genode::Lock>::Lock_guard (lock=..., this=<synthetic pointer>) at .../repos/base/include/base/lock_guard.h:40
#6  Libc::Vfs_plugin::select (this=0x13e5580 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+640>, nfds=<optimized out>, readfds=0x7f99e8dff480, writefds=0x7f99e8dff500, exceptfds=<optimized out>, timeout=<optimized out>) at .../repos/libports/src/lib/libc/vfs_plugin.cc:1230
[... cut ...]
#9  0x0000000001365188 in select (nfds=<optimized out>, readfds=readfds@entry=0x7f99e8dffa30, writefds=writefds@entry=0x7f99e8dffab0, exceptfds=exceptfds@entry=0x7f99e8dffb30, timeout=timeout@entry=0x7f99e8dffa20) at .../repos/libports/src/lib/libc/select.cc:307
#10 0x0000000001362fa1 in poll (fds=0x7f99e4c18788, nfds=1, timeout=<optimized out>) at .../repos/libports/src/lib/libc/poll.cc:84
...
```

In frame `#8`, the select callback list lock is acquired. In frame `#6`, the `Libc::Vfs_plugin::select()' function waits for the vfs lock.

chelmuth, 2020-08-13:
Commit f2c6b15a05d2d8138ef679f617ff02883aba2fe4 above addresses this issue by implementing the monitor (thread) pattern for the VFS instance in libc.

@nfeske please review the changes
@cproc please rebase your work on top of this implementation as it reenables the update of file timestamps

chelmuth, 2020-08-13:
The current implementation executes the monitors in a dedicated I/O signal handler. This may result a deadlock depicted by the following backtrace (in this case due to a nested call to `wait_and_dispatch_one_io_signal()`). It seems the monitor execution  in the libc kernel must be moved after the dispatch function returned because of the generated side effects.

```
#1  0x0000000050052688 in lx_futex (val=0, op=0, uaddr=<optimized out>) at .../repos/base-linux/src/lib/syscall/linux_syscalls.h:441
#2  thread_stop_myself (myself=<optimized out>) at .../repos/base-linux/src/include/base/internal/lock_helper.h:62
#3  Genode::Lock::lock (this=this@entry=0xaff8c38 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+16176>, myself=...) at .../repos/base/src/lib/base/lock.cc:128
#4  0x0000000050052baf in Genode::Mutex::acquire (this=this@entry=0xaff8c38 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+16176>) at .../repos/base/src/lib/base/mutex.cc:25
#5  0x0000000050057bbf in Genode::Mutex::Guard::Guard (mutex=..., this=<synthetic pointer>) at .../repos/base/include/base/mutex.h:62
#6  Genode::Registry_base::_for_each(Genode::Registry_base::Untyped_functor&) () at .../repos/base/src/lib/base/registry.cc:109
#7  0x000000000af4f740 in Genode::Registry<Libc::Monitor::Job>::for_each<Libc::Monitor::Pool::execute_monitors()::{lambda(Libc::Monitor::Job&)#1}>(Libc::Monitor::Pool::execute_monitors()::{lambda(Libc::Monitor::Job&)#1} const&) (fn=..., this=<optimized out>) at .../repos/base/include/util/interface.h:20
#8  Libc::Monitor::Pool::execute_monitors (this=<optimized out>) at .../repos/libports/src/lib/libc/internal/monitor.h:149
#9  Libc::Kernel::_monitors_handler (this=<optimized out>) at .../repos/libports/src/lib/libc/internal/kernel.h:286
#10 0x00000000500490c9 in Genode::Entrypoint::_wait_and_dispatch_one_io_signal(bool) () at .../repos/base/src/lib/base/entrypoint.cc:197
#11 0x00000000014e4ec1 in Genode::Entrypoint::wait_and_dispatch_one_io_signal (this=<optimized out>) at .../repos/base/include/base/entrypoint.h:201
#12 Lx::Timer::wait (timeo=994, this=0x151d880 <Lx::timer_init(Genode::Entrypoint&, Genode::Timeout_scheduler&, Genode::Allocator&, void (*)())::inst>) at .../repos/dde_linux/src/lib/lxip/timer_handler.cc:312
#13 schedule_timeout (timeout=timeout@entry=994) at .../repos/dde_linux/src/lib/lxip/timer_handler.cc:403
#14 0x0000000001472f1d in sk_stream_wait_memory (sk=sk@entry=0x7ffff40cb040, timeo_p=timeo_p@entry=0x401fea88) at .../contrib/dde_linux-173719007927a108a9558e727b8c9b388f3159a2/src/lib/lxip/net/core/stream.c:148
#15 0x00000000014a6c90 in tcp_sendmsg (sk=0x7ffff40cb040, msg=<optimized out>, size=<optimized out>) at .../contrib/dde_linux-173719007927a108a9558e727b8c9b388f3159a2/src/lib/lxip/net/ipv4/tcp.c:1274
#16 0x000000000ad1f9c7 in Vfs::Lxip_data_file::write (this=0x7ffff7fe74c0, src=<optimized out>, len=4194304) at .../repos/dde_linux/src/lib/vfs/lxip/vfs.cc:496
#17 0x000000000ad1e3ab in Vfs::Lxip_vfs_file_handle::write (this=<optimized out>, src=<optimized out>, count=<optimized out>, out_count=@0x401febb8: 0) at .../repos/os/include/vfs/vfs_handle.h:129
#18 0x000000000ad1eb6d in non-virtual thunk to Vfs::Lxip_file_system::write(Vfs::Vfs_handle*, char const*, unsigned long long, unsigned long long&) () at .../repos/dde_linux/src/lib/vfs/lxip/vfs.cc:1970
#19 0x000000000af8d0e2 in Libc::Vfs_plugin::<lambda()>::operator() (__closure=0x403fec50) at .../repos/libports/src/lib/libc/vfs_plugin.cc:887
#20 Libc::Monitor::Result Libc::Monitor::monitor<Libc::Vfs_plugin::write(Libc::File_descriptor*, void const*, unsigned long)::{lambda()#3}>(Genode::Mutex&, Libc::Vfs_plugin::write(Libc::File_descriptor*, void const*, unsigned long)::{lambda()#3} const&, unsigned long long)::_Function::execute() () at .../repos/libports/src/lib/libc/internal/monitor.h:82
#21 0x000000000af4fe26 in Libc::Monitor::Job::execute (this=0x403fea40) at .../repos/libports/src/lib/libc/internal/monitor.h:111
#22 Libc::Monitor::Pool::execute_monitors()::{lambda(Libc::Monitor::Job&)#1}::operator()(Libc::Monitor::Job&) const (__closure=<optimized out>, job=...) at .../repos/libports/src/lib/libc/internal/monitor.h:150
#23 Libc::Monitor::Pool::execute_monitors()::{lambda(Libc::Monitor::Job&)#1}::operator()(Libc::Monitor::Job&) const (job=..., __closure=<optimized out>) at .../repos/libports/src/lib/libc/internal/monitor.h:149
#24 void Genode::Registry<Libc::Monitor::Job>::for_each<Libc::Monitor::Pool::execute_monitors()::{lambda(Libc::Monitor::Job&)#1}>(Libc::Monitor::Pool::execute_monitors()::{lambda(Libc::Monitor::Job&)#1} const&)::Typed_functor::call(void*) (this=<optimized out>, obj_ptr=0x403fea40) at .../repos/base/include/base/registry.h:123
#25 0x0000000050057c86 in Genode::Registry_base::_for_each(Genode::Registry_base::Untyped_functor&) () at .../repos/base/src/lib/base/registry.cc:134
#26 0x000000000af4f740 in Genode::Registry<Libc::Monitor::Job>::for_each<Libc::Monitor::Pool::execute_monitors()::{lambda(Libc::Monitor::Job&)#1}>(Libc::Monitor::Pool::execute_monitors()::{lambda(Libc::Monitor::Job&)#1} const&) (fn=..., this=<optimized out>) at .../repos/base/include/util/interface.h:20
#27 Libc::Monitor::Pool::execute_monitors (this=<optimized out>) at .../repos/libports/src/lib/libc/internal/monitor.h:149
#28 Libc::Kernel::_monitors_handler (this=<optimized out>) at .../repos/libports/src/lib/libc/internal/kernel.h:286
#29 0x00000000500490c9 in Genode::Entrypoint::_wait_and_dispatch_one_io_signal(bool) () at .../repos/base/src/lib/base/entrypoint.cc:197
#30 0x000000000af55dd5 in Genode::Entrypoint::wait_and_dispatch_one_io_signal (this=<optimized out>) at .../repos/base/include/base/entrypoint.h:201
```

nfeske, 2020-08-17:
With the current version on staging, the ldso test fails right at the end of the test.

```
....
[init -> test -> test-ldso] ~Lib_2_global 0x11223340
[init -> test -> test-ldso] Warning: rtc not configured, returning 0
[init -> test -> test-ldso] Error: libc suspend() called from non-user context (0xaf58a7c) - aborting
```

The backtrace reveals the following:
```
#7  0x000000000af81b4e in Libc::Monitor::monitor<Libc::Vfs_plugin::close(Libc::File_descriptor*)::<lambda()> > (timeout_ms=0, fn=..., mutex=..., 
    this=0xaff4d10 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+40>)
    at .../repos/base/include/util/interface.h:20
#8  Libc::Vfs_plugin::close(Libc::File_descriptor*) () at .../repos/libports/src/lib/libc/vfs_plugin.cc:585
#9  0x000000000af25734 in __sys_close (libc_fd=<optimized out>) at .../repos/libports/src/lib/libc/file_operations.cc:247
#10 0x000000000af481c9 in close_file_descriptors_on_exit () at .../repos/libports/src/lib/libc/kernel.cc:500
#11 0x00000000500aa47b in genode___cxa_finalize (dso=dso@entry=0x0) at .../repos/base/src/lib/startup/_main.cc:166
#12 0x00000000500a5437 in __cxa_finalize (dso=dso@entry=0x0) at .../repos/base/src/lib/cxx/misc.cc:66
#13 0x00000000500aa533 in genode_exit (status=status@entry=123) at .../repos/base/src/lib/startup/_main.cc:188
#14 0x000000000af1ed38 in _exit (status=status@entry=123) at .../repos/libports/src/lib/libc/exit.cc:24
#15 0x000000000af1ed86 in exit (status=status@entry=123) at .../repos/libports/src/lib/libc/exit.cc:42
#16 0x00000000010022df in Libc::Component::construct (env=...) at .../repos/libports/src/test/ldso/main.cc:280
```

The culprit is that the atexit handling still resides in Genode's base lib. It must be moved to the libc.

nfeske, 2020-08-18:
Commit 57f38dc3778aee7be9baaad4bdd4ad8531d183ec fixes a problem with `Libc::Component` cases such as the ssh_terminal (ssh_exec_channel.run). Monitors must not depend on the `_io_ready` flag because this flag is only set by VFS operations, not the monitors signal handler.

commit f2c6b15a (2020-08-13): libc: process VFS operations in monitor

Fixes #2635
commit a115ebc9 (2020-08-14): libc: process VFS operations in monitor

Fixes #2635
closed 2020-08-25 by nfeske