genodelabs/genode #2637: nic_router: cannot re-use a "domain" session

skalk, 2018-01-09:
Yesterday, I could repeatedly reproduce the following problem:

I have several VMs that are routed to different "domains" of the nic_router. I restarted a VM, which had a working network setup in the first place. After restarting the VM, I still get a DHCP response from the nic_router (including DNS settings), but no network requests are handled apart from that. I tried to connect to the given DNS server and directly to an IP address of the local network, both did not work. After a fresh cold host reboot, I could reproduce the problem with two VMs that got a working network setup after a first boot, but not after a reboot. A constantly running VM, which did not reboot, always had network access through the nic_router at all times.

m-stein, 2018-01-09:
I'll have a look. Thanks for the description :)

skalk, 2018-01-19:
@m-stein did you really destroyed the subsystem containing Virtualbox or did you reboot within the virtual machine? I expect it can be reproduced by re-starting Virtualbox only, not be rebooting Linux inside.

m-stein, 2018-01-19:
@skalk This should do the trick:
7c4a855 nic_router: fix use of outdated ARP-cache entries

Does it fix your problems?

skalk, 2018-01-22:
@m-stein at least after restarting the VM I'm using for web-browsing just now, I'm able to write by re-using the nic_router domain of it. Thank you very much for looking into this!

nfeske, 2018-01-22:
Merged to staging.

commit 552b0a67 (2018-01-22): nic_router: fix use of outdated ARP-cache entries

When a NIC session is destructed at the router, we have to remove all ARP
cache entries that match the MAC address of that session. Otherwise the
outdated entries might be re-applied later, leading to wrong destination
MAC addresses in routed packets.

Fixes #2637
closed 2018-02-09 by nfeske