genodelabs/genode #2642: VFS Rump file descriptor leak

ehmry, 2018-01-11:
The Rump VFS plugin is leaking file descriptors as evidenced by **EMFILE** errors after heavy usage. I can see that descriptors may leak when Rump allocates a descriptor but a VFS file handle fails to allocate. I will first fix that, and then see if I can get the VFS to catch all `Out_of_ram` and `Out_of_caps` exceptions to return error values rather than let exceptions propagate into the application.

ehmry, 2018-01-12:
Luckily this **EMFILE** error only occurs on my #1934 topic branch, and staging is unaffected. I do however have four commits for catching leaks during handle allocation on my `issue2642_vfs_leaks` branch.

ehmry, 2018-01-12:
Looking back at the patches I think it would be wise to make the File_system `Out_of_ram` and `Out_of_caps` exceptions distinct from the Genode exceptions. I would like to have a decision on this before the rest of the patches.

chelmuth, 2018-01-17:
This is quite the opposite of

  4d442bc Streamline exception types

Please elaborate more on your motivation to reintroduce special exception types.

ehmry, 2018-01-18:
On further consideration, I don't think special types are necessary. My motivation was to distinguish remote exceptions from local exceptions, but these out_of_resource exceptions are caught directly from RPC functions in File_system::Connection, so I see no ambiguity.

ehmry, 2018-01-26:
Rebased against staging, without changes to exceptions classes, at my *issue2642_vfs_leaks* branch

nfeske, 2018-02-10:
Thanks @ehmry, I merged the commits to staging now.

chelmuth, 2018-03-01:
Is this issue fixed by the merged commits or are there open points?

ehmry, 2018-03-02:
Yes, fixed in master.

closed 2018-03-02 by ehmry