genodelabs/genode #2644: base/xml_node: corner case of dst_len = 1 in Xml_node::decoded_content

BorisMulder-CSL, 2018-01-15:
in the Xml_node class (https://github.com/genodelabs/genode/blob/staging/repos/base/include/util/xml_node.h), the decoded_content function does not copy over a string of exactly 1 character (see line 686). This can cause faults in code using this function. 

Unless there is a reason to why this is done so, may I suggest changing line 686
` for (; dst_len > 1 && src_len; result_len++) {`
into
` for (; dst_len > 0 && src_len; result_len++) {`

m-stein, 2018-01-15:
Hi Boris,

As far as I can tell, the implementation of [1], where [2] is called,
indicates that the destination buffer is considered to be
zero-terminated. This explains why the loop skips copying with a buffer
size of 1. Though it remains questionable wether the code would become
more descriptive if we moved the zero-termination to [2].

[1]
template <typename STRING>
STRING decoded_content() const

[2]
size_t decoded_content(char *dst, size_t dst_len) const

nfeske, 2018-01-16:
@BorisMulder-CSL Thank you very much for having a close look. You spotted one problem (A) and made @chelmuth spot another (B) while looking at the code! Regarding (A), the non-STRING version of `decoded_content` is not expected to zero-terminate the buffer. This is the responsibility of the caller. E.g., the STRING version does that. So Boris is right in that the check should be changed to `dst_len > 0`. However, the more embarrassing problem (B) is that the code apparently misses to decrement `dst_len` in each iteration. So the destination buffer may overflow if `src_len` is larger than `dst_len`. (at least from looking at the code, we should add a test to our test/xml_node. @m-stein Would you like to try this out?)

m-stein, 2018-01-16:
@nfeske I'm at it.

m-stein, 2018-01-16:
My suggestion:
625d754 Xml_node: fix bug in Xml_node::decoded_content
b5174f6 Xml_node: do not consider '0' in decoded_content
cb29a3b test/xml_node: test Xml_node::decoded_content

nfeske, 2018-01-16:
@m-stein Thank you very much!

m-stein, 2018-01-18:
@nfeske Thanks for the hint:
059dddd fixup "test/xml_node: test Xml_node::decoded_content"

nfeske, 2018-02-09:
Fixed in master.

closed 2018-02-09 by nfeske