genodelabs/genode #2654: trace_logger: convenient tracing frontend

m-stein, 2018-01-24:


m-stein, 2018-01-24:
This is my suggestion for a basis of what could become a general tool for gathering, processing, and exporting tracing data in a convenient way.

083d3ba trace_logger: convenient tracing frontend
92d32a0 trace/policy/rpc_name: enable building again
b325527 AVL node/tree: make non-copyable
da4c23c trace_session: fix exception list in 'trace' RPC
ec82a31 base/trace/buffer.h: fix circular #include
437e692 create_uboot: better documentation
ff365e8 init config.xsd: add ld_verbose attribute

m-stein, 2018-01-24:
Features I'd like to add in the future:
* Introduce domain tag similar to the NIC router configuration - So far, it is not possible to re-use one configuration for multiple policies.
* More systematic and descriptive names for configuration attributes
* Enable different output formats (log, report, FS ...) - I'm not sure with this one as there exists a variety of converter components that might do the job as well (then, the question would be which output format to use in the first place).
* Automatic test for the functionality of the trace_logger (current run-script is a mere demo)
* Gather more types of data (for instance RAM/CAP quota consumption)

nfeske, 2018-01-30:
Thanks for this great commit series! I very much appreciate the great care that went into the trace_logger, in particular the clarity of the code, documentation, and the illustrative run script.

I merged the commits to staging.

Regarding your further ideas, I would wait with adding usability features before the tool has seen a good deal of use. I suggest to let's first see how far we get with the current version. Outputting the buffer to the LOG is fine as far as I can see. I support the idea to upgrade the run script to become an automated test. Regarding further events to capture, here are my ideas:

* Packet-stream operations (maybe capturing the number of packets that are in-flight to get an idea how well packets are batched?)
* The moment when a thread gets blocked on because of lock contention, and the moment when a thread gets unblocked when aquiring a contented lock.
* An easier way to insert manual instrumentations. Right now, manual trace entries can be generated by the `Thread::trace` function that takes a character pointer as argument, which is rather cumbersome. It would be cool to add trace-buffer entries with the same ease as we insert 'log' messages. E.g., how about adding a new `trace` function that works analogously to `log`, `warning`, `raw`, and `error`?

m-stein, 2018-01-30:
@nfeske Your suggestions sound pretty cool to me, especially writing to the buffer with something alike Genode::log.

m-stein, 2018-02-07:
@chelmuth In these commits the compile errors we had recently with b325527 should be fixed:
46cee2e trace_subject_reporter: fix missing include
65ae762 AVL node/tree: make non-copyable

chelmuth, 2018-02-07:
What about the following targets?
```
- arndale: make LIB=core-hw
- arndale: make server/vmm
```

m-stein, 2018-02-07:
@chelmuth I'm at it.

m-stein, 2018-02-07:
@chelmuth Sorry for the inconvenience:
a890b94 trace_subject_reporter: fix missing include
2704697 AVL node/tree: make non-copyable

commit 99a5324b (2018-01-30): trace_logger: convenient tracing frontend

The 'trace_logger' component can be used to easily gather, process and export
different types of tracing data. Which subjects to select is configurable via
session label policies and thread names. Which data to collect from the
selected subjects can be configured for each subject individually, for groups
of subjects, or for all subjects. The gathered data can be exported as log
output.

This is an example configuration of the 'trace_logger' component which shows
the default value for each attribute except the policy.thread and
policy.label:

! <config [... cut]
closed 2018-02-09 by nfeske