genodelabs/genode #2657: `uname -a` displays odd characters

ehmry, 2018-01-26:
The uname utility from coreutils expects all string buffers processed by the `sysctl` call to be zero-terminated, even if `sysctl` returns a zero-length result. WIthout placing a zero at the front of the client buffer `uname` will display artifacts from the buffer.

ehmry, 2018-01-26:
Fixed at ae951ec27624bc79032f25b36e2d8215c8ae8c3b.

nfeske, 2018-01-29:
Thanks for looking into it. Would you mind assigning the null termination via `Genode::strncpy` with src="" and size=1 ? This will implicitly handle the case where the destination-buffer length is zero. Or can we rule out this corner case otherwise?

ehmry, 2018-01-29:
This corner case is handled earlier, but I after some more tests I think it is better to just to zero the whole buffer before handling any cases. New patch at c77991294765725347177dd7e0605298309a3c7a.

nfeske, 2018-01-30:
All the better! Thank you. Merged to staging.

commit ca507031 (2018-01-30): Libc: zero sysctl buffer before strncpy

The uname utility from coreutils needs its read buffer zero-terminated,
regardless of read length.

Fix #2657
closed 2018-02-09 by nfeske