genodelabs/genode #2659: foc: parent gets stuck when killing a child as direct response to `Parent::exit` RPC call

nfeske, 2018-01-30:
On Genode, there are spurious (harmless but ugly) page-fault messages at child-destruction time that result from the sequence how the child's sessions are closed. I.e., if the child's PD and binary-ROM sessions are closed before the child's CPU session, the child's threads are still scheduled by the kernel but lack any valid virtual-memory mappings. To reduce the noise in the log, I refined the child destruction such that the CPU sessions of the to-be-killed child are closed prior all other sessions. There are two events when this code is executed. First, when the parent decides to kill a child (e.g., as response of a init configuration change). Second, when the child calls the 'Parent::exit' RPC function.

The latter is the situation where things go wrong on Fiasco.OC. Apparently, the parent gets stuck as soon as the child's CPU session is destructed, which is unexpected.

One possible (but unproven) explanation would be that the parent runs on the time slice of the to-be-destructed child while the child is killed. So with regard to scheduling, there exists a dependency of the parent from the child. It seems like the kernel misses to dissolve this dependency when the child is destructed, never scheduling the kernel again. However, when delaying the child destruction for a bit, the problem seems to disappear (maybe because the parent runs on its own time slice at that time).

alex-ab, 2018-01-30:
Similar context (just pthread) as in #1495

nfeske, 2018-01-30:
@alex-ab Thanks, that is interesting.

With the fixup 34eaae346a7e7bc3d11b65f85e6f43b9acdcbbf8 I work around this issue to make the autopilot happy. It is just a cover-up. I must confess that I am not eager to debug this. The underlying problem stays unresolved for now.

nfeske, 2018-03-01:
Since we don't plan to address the kernel problem and the above work-around has entered the master branch, I am closing this issue.

closed 2018-03-01 by nfeske