genodelabs/genode #2664: vfs: starvation of file_system session adapter on heavy writes

alex-ab, 2018-02-01:


chelmuth, 2018-02-01:
What a catch! Does this fix the VM install corruption @skalk and you experienced?

alex-ab, 2018-02-01:
Partly, I'm still investigating. The commits solve the issue that the guest complains about a stuck device model and trying to reset it and finally to fail after some attempts. This typically happens during installation and/or download of new packages within the guest.

I have now a setup where the guest content (growing vdi) is written to two file system servers. On a read the delivered content is compared between both. Unfortunately, they disagree sometimes ...

chelmuth, 2018-02-01:
Hui, this sounds like lock-stepping file systems. Thanks, for your efforts!

alex-ab, 2018-02-12:
Set to fixed, commits fixed starvation as far as I can tell.

nfeske, 2018-02-12:
Great! I merged them to staging just now.

closed 2018-02-16 by alex-ab