genodelabs/genode #2668: wifi_drv: generate report on authentication failure

BorisMulder-CSL, 2018-02-02:
For the wifi_drv, it would be nice if it would generate a report if it failed to connect to the network for instance when the password is incorrect. It does give a quite verbose log output, but reporting this would be nice so another component could for example ask the user to re-enter the password.

Right now it only reports state when it is connected or disconnected, and very rarely it gives a disconnect report when it has tried to connect for multiple times.

I also wonder what logic is behind the retries. When I fill in an incorrect password, sometimes it tries to connect forever and sometimes it gives up really quickly. Also, sometimes it generates a disconnected state report and sometimes it does not. Is there any logic behind this?

cnuke, 2018-02-06:
That would be a nice feature indeed, patches are welcome.

To answer your second question, which also influnces the current behaviour regarding connecting state reporting, it depends on the WPA supplicant application. If you change the wlan configuration, the supplicant might try to authenticate/associate with a network serveral times and it is up to its internal heuristics to decide how many attempts it will make before it adds the network to its blacklist and stops trying for some time. Since we also rely on the supplicant for generating the state reports, we only get one if it is in a state where it will generate one.

The supplicant is still treated like a blackbox and eventually it would be better to have more control over it but we do not use the proper control interface that would enable us to do so yet.

BorisMulder-CSL, 2018-02-06:
Well, I already tried to add it but it does not really work as intended this way.
What I did was changing the reporting in the wpa_report_disconnect_event to the following:
```
+			unsigned auth_failures = wpa_ssid->auth_failures;
+
 			xml.node("accesspoint", [&]() {
 				xml.attribute("ssid", ssid.string());
 				xml.attribute("bssid", bssid_buf);
 				xml.attribute("state", "disconnected");
+				xml.attribute("auth_failures", auth_failures);
 			});
```
However, the times when this report is generated seem to be weird sometimes. For now this report is sometimes generated after a few tries, and sometimes it is never generated if the password is incorrect and it keeps trying to connect with the wrong password forever.
I also noticed that sometimes the report is generated even though the entered password is correct. This often happens when the password is stored persistently and it tries to connect again after a reboot of the system. Whenever I start the wifi_drv and update the wlan_configuration, it usually connects immediately, but it randomly sometimes never manages to authenticate and keeps producing auth_failure log messages and reports.
Again, the time after this report is produced seems very random, if it is produced at all.

BorisMulder-CSL, 2018-02-06:
Here is the commit:
https://github.com/nlcsl/genode/commit/54ef3c876179740b4b4ccd766b18607353ef3597

commit 964aac59 (2018-03-10): wpa_supplicant: report auth_failures

Fixes #2668
closed 2018-03-29 by chelmuth