genodelabs/genode #2669: WIP libgfxinit based mode-setting driver

cnuke, 2018-02-06:
This is a WIP port of libgfxinit (a library written in Ada/SPARK that configures the display engine of Intel HD Graphics GPUs to provide basic mode-setting and is used by the coreboot firmware) and its dependencies, namely a small Ada runtime and libhwbase. So far it was only tested with the internal display of a Thinkpad X250.

The Genode build system already provides Ada integration through the use of gnatmake. By now this tool is marked as deprecated and its use is discouraged in favor of gprbuild by Ada's upstream development. Since using gnatmake turned out to be unfitting to use anyway while doing the port, it was replaced by a hand-rolled compile mechanism. Unfortunately it lacks dependency tracking, which makes compiling Ada code somewhat cumbersome though.

I think the most reasonable way forward would be to use of gprbuild from within our build system but I am not sure how to proceed in this matter (AFAIU gprbuild relies on existing information about the target system). I am curious how Componolit currently integrates their Ada based components. Maybe @senier or @jklmnn could comment on that?

That being said, just cherry-pick the commits below and build the demo scenario to try it out. For now each Ada based library has to be build several times to pick up all generated files (*.ali and b__*). So invoking `make LIB=ada_rts` a few times (the some goes for `libhwbase` and `libgfxinit`) in that order before building the rest of the scenario works best.

senier, 2018-02-06:
We use the existing `gnatmake` support so far. As we are forced to have all source in one place, properly supporting Ada programs and libraries in Genode is high on our TODO list. Can you elaborate why `gnatmake` did not fit for what you were trying to do?

Eventually we may want to use `gprbuild`. However, an integration into the Genode build system could be challenging, as both systems try to be smart about build dependencies and consider themselves as the master of the build process. As `gprbuild` mostly relies on config files, we won't be able to configure everything on the command line as with gnatmake and other tools. We would probably need to generate a project file from the `target.mk` settings as well as a toolchain configuration that forces `gprbuild` to use the Genode toolchain. Not sure whether this would work.

I'm a bit skeptical about getting your hand-crafted build approach right.

jklmnn, 2018-02-06:
I've seen that you included the runtime within your commits. @senier and I have already discussed different options and as far as I remember we came to the conclusion that it probably would be better to add the runtime as a port and not as code to Genode.

First it isn't necessary to copy the code and make updates of the runtime at least inconvenient. Second different components may require different levels of runtime complexity and support and the smallest component shouldn't depend on all the same things as the most complex one.

I had a concept similar to the dde_linux file lists in mind where you select the parts of the runtime you need with two or three predefined runtimes (such as different selections from the same base runtime).

Beside that I'm aware that this is WIP and that the concepts I described require more effort before yielding usable results.

cnuke, 2018-02-07:
Thanks for your insights.

@senier okay, good to know. To be honest, you have me at a loss here as I merely remember that I had problems to get `gnatmake` to work with spliting the component up in seperate libraries (PEBKAC not excluded). Building it all-in-one worked (aside from the missing link between the Genode build system and `gnatmake` wrt to dependency tracking). My somewhat futile attempt was to counter the problems but I am skeptical as well, hence my initial question.

@jklmnn I agree, the runtime files should not pollute the Genode repository unnecessarily and the commit is only there out of laziness^Wconvience as you have already guessed.

jklmnn, 2018-05-04:
As you might have noticed runtime and library support is available on master now (https://github.com/genodelabs/genode/pull/2748). I also issued https://github.com/genodelabs/genode/pull/2810 which adds secondary stack support and makes it easier to use additional parts of the runtime (either from the gcc itself or custom implementations).

cnuke, 2018-05-04:
Yes, thanks for the notification - I followed the issue. Adapting the libgfxinit port etc. is on my to-do list but I won't have the time to do it in the foreseeable future. If you would like to get your hands dirty, be my guest.

nfeske, 2021-02-26:
There is no plan to pick this driver up for mainline. Hence, I'm closing this stale issue.

closed 2021-02-26 by nfeske