genodelabs/genode #2670: NIC router: dynamic reconfiguration

m-stein, 2018-02-06:


m-stein, 2018-02-06:
Might be interesting for people that use the router in Sculpt:
88c31eb nic_router: make verbosity reconfigurable
f7eff31 nic_router: install empty configuration handler
ead76de nic_router: conform domain labeling in log
07e1278 nic_router: add verbose_packets per domain
32fd70a nic_router: add verbose_packets attribute

But so far, it's all "experimental" because the last commits implies a small feature regression.

m-stein, 2018-02-07:
Two additional commits that fix the feature regression:
0328014 nic_router: rename Domain_avl_member & Domain_base
d6eb218 Fixup "nic_router: make verbosity reconfigurable" (invalid domains)

m-stein, 2018-02-07:
I managed to attach interfaces to their domain dynamically. So, an interface can also be detached and is then in a suspended state until it is attached again. I assume that I must also change the Nic_session::link_state of a suspended interface to "down" in order that NIC clients redo for instance DHCP (the DHCP role of the domain might have changed) when the interface gets unsuspended again.

nfeske, 2018-02-07:
That sounds fantastic.

I'm curious, does this also work for the uplink? If yes, we'd have now a way to replace the wifi driver by the nic driver on the fly: First, remove the uplink from the nic router, second, replace nic driver (which does not bother the nic bridge since it has no longer a nic session to the driver), third re-create the uplink in the nic-router config. Is this realistic?

m-stein, 2018-02-08:
The mechanism is generic for all interfaces (including uplink) except the NIC session link state of course. But I have to do further modifications (automatic attach/detach when domains appear/disappear) and testing to give you a definite answer.

m-stein, 2018-02-09:
I now can add a domain at runtime and get a routed connection between one session that was waiting for this domain and another that was already attached to a domain.

m-stein, 2018-03-19:
Fixed by:
d16fed2 nic_router_dyn_config.run: test re-configurability
0a2ff7f nic_router: handle configuration changes
2d0ed7c nic_router: destroy list items during for_each
11cde99 nic_router: != operator for IP config
c24b271 nic_router: != operator for IP address prefix
634be66 nic_router: overwritable reference wrapper
55a5659 nic_router: alloc specific bits at bit allocator
b3cd5f1 nic_router: alloc specific port at port allocators
ea99d5c nic_router: safe pointer class for const objects
3e4d03b nic_router: Avl_tree wrapper with destroy_each
e95135f nic_router: destroy_each for domain tree
e96b4b0 nic_router: destroy_each for list wrapper
573b2b4 nic_router: safe pointers that are valid and const
97027ca nic_router: support interfaces without a domain
1b7cc75 nic_router: conform domain labeling in log
bf0d1e2 nic_router: add verbose_packets per domain
419514a nic_router: add verbose_packets attribute
d213f02 nic_router.run: test DHCP server
b7b3df6 test/lwip/udp/client: deal with packet loss
281ded6 list: warn on double insertions (suggestion)
0bdf207 bit_allocator: alloc specific block of bits
a7dc8fd trace_subject_reporter: fix missing #include
e6434bf test/lwip/http_clnt: don't succeed falsely
7947727 util/string: add ascii_to for unsigned short
072e4a1 list_model: fix missing include

Missing:
1) An automatic stress test for the re-configurability
2) A README update

I would add the missing points soon. I plan to do number 1 in the context of a complete new NIC router test that replaces the currently separate tests (routing, reconfiguration, DHCP) into one big test. I think the missing points are not critical for merging.

nfeske, 2018-03-20:
I like the commit series.

Regarding the safety check in `list.h`, I would leave it out for now. Even though I see the potential benefit of the safety check, the `List` is unsafe in the first place and should be avoided by application code. The attempt to make it safe is futile while making it slower. It should serve merely as a building block for a safer data structure, as done by `List_model` and `Registry`. I also prefer to not include `base/log.h` from the `util/list.h` header.

Speaking of lists, have you considered using the `Registry` utility or have you found it too limiting for your uses?

m-stein, 2018-03-20:
@nfeske Thanks for the review and the fast feedback! Regarding the ``Registry`` utility - When creating the router I had no notion of this utility and now I was concentrating on the re-configuration feature and did not consider replacing existing datastructures. I would treat this as a new issue.

nfeske, 2018-03-20:
No problem. I am not urging you but was just curious. If we are lucky, the `Registry` may make your list wrapper obsolete. If not, we can take your use case as a new requirement. ;-)

m-stein, 2018-03-20:
Updated the commits according to the suggestions:
367e3cf nic_router_dyn_config.run: test re-configurability
a1e8477 nic_router: handle configuration changes
20b3e3b nic_router: destroy list items during for_each
17e339d nic_router: != operator for IP config
10454e6 nic_router: != operator for IP address prefix
f15f797 nic_router: overwritable reference wrapper
a1d292a nic_router: alloc specific bits at bit allocator
b0f8a46 nic_router: alloc specific port at port allocators
43675d1 nic_router: safe pointer class for const objects
33de87e nic_router: Avl_tree wrapper with destroy_each
1643aef nic_router: destroy_each for domain tree
cf29262 nic_router: destroy_each for list wrapper
9fb6dee nic_router: safe pointers that are valid and const
6882c07 nic_router: support interfaces without a domain
6c70d2e nic_router: conform domain labeling in log
0f985c8 nic_router: add verbose_packets per domain
3cd103b nic_router: add verbose_packets attribute
b314b32 nic_router.run: test DHCP server
2a1d4e2 test/lwip/udp/client: deal with packet loss
0bdf207 bit_allocator: alloc specific block of bits
a7dc8fd trace_subject_reporter: fix missing #include
e6434bf test/lwip/http_clnt: don't succeed falsely
7947727 util/string: add ascii_to for unsigned short
072e4a1 list_model: fix missing include

m-stein, 2018-04-03:
Additional commits:
3bfacec nic_router: conforme log in Domain
d78f5e6 nic_router: fix uncaught Domain_tree::No_match

closed 2018-04-10 by chelmuth