genodelabs/genode #2672: vfs: file_system session adapter (client) drops silently data on partial (server) writes

alex-ab, 2018-02-06:


alex-ab, 2018-02-06:
The original [code](https://github.com/genodelabs/genode/commit/b0935ef9b27edbb74da8d7c7345d78e5e22119d5#diff-8309a9a3742915d2bd51d59b4a46c51aL220) has no appropriate equivalent, e.g. in the [write ack](https://github.com/genodelabs/genode/commit/b0935ef9b27edbb74da8d7c7345d78e5e22119d5#diff-8309a9a3742915d2bd51d59b4a46c51aR525) the actual written length would need evaluation, but it seems there are no metadata (the original size) to compare to.

ehmry, 2018-02-07:
One solution would be to set an error flag on the file handle and return something like ERR_IO from sync.

ehmry, 2018-02-08:
I could implement forwarding errors to the next sync in a short amount of time, in fact I may already have a patch that does this.

alex-ab, 2018-02-12:
0e29cbfe9c9d974bf498a7404c472b31d923bd16 showcases the bug. I adjusted all file system server (I'm aware of) to not acknowledge a succeeded packet if the written length is not the expected one. Already the ram_fs server returns errors for the libc_vfs.run test.

alex-ab, 2018-02-13:
Updated it according to the offline discussion.

nfeske, 2018-02-13:
Very good. Would you like to keep the individual "handle result propagation better" commits or should we squash them to one? (I'd like to add a few words of background info in the commit message)

alex-ab, 2018-02-13:
I'm fine with squashing.

nfeske, 2018-02-13:
Merged to staging. @alex-ab are you fine with my commit message?

alex-ab, 2018-02-13:
Yes, thanks!

closed 2018-02-16 by alex-ab