genodelabs/genode #2676: Sculpt for Early Adopters (EA)

nfeske, 2018-02-11:
Genode 18.02 will feature the first variant of the Sculpt scenario, which enables the use of Genode as rudimentary desktop OS. For the big picture behind this development, please refer to the road map: https://genode.org/about/road-map

Right now, Sculpt resides on a feature branch (https://github.com/nfeske/genode/commits/sculpt) along with the branch `sculpt_next` for the next iteration. It already covers most functionality intended for Sculpt EA:

* Base system with the drivers and components needed for an interactive system
* A way for tweaking and monitoring the system at runtime (leitzentrale)
* Support for wifi and wired networking
* Support for preparing the disk (block runtime)
* Support for accessing files stored on the disk (fs runtime)
* Support for restoring saved settings (load runtime)
* Installing software (update runtime)
* Deploying installed software (deploy runtime)
* VirtualBox (including networking and shared folders)
* USB-passthrough to VirtualBox

The following points are still subject to change:

* At the current feature branch, VirtualBox is still included in the base image (vm runtime). In the release, I only want to offer it as installable software package (deploy runtime).
* The current deploy runtime contains several things that should better be deferred to Sculpt TC (in May). I.e., the window manager and PDF viewer. On Sculpt EA, I merely want to offer the components needed for the basic VM scenario. The other packages will be available, but just as a preview to TC.
* I'd leave out audio support and acpica from Sculpt EA. For venturesome people, the packages will be available, but not by default. The main reason is that there are known issues, which may leave a bad impression if we advertised them now. E.g., acpica does not update the reports after the initial one, or the audio out is choppy when using certain guests in VirtualBox.
* Right now, the block device accessed by the default file system is manually defined in the boot image (ahci-0, ahci-1), which prevents us to use one ISO image across all machines. To overcome this situation, I plan the following change: Whenever the driver manager detects only one block device, it makes this device available under the label "default". So in the `subinit/default_fs.config`, we can always refer to the `default` block device. This will hopefully capture the situation of most users who only use one SATA disk. In other situations (also when using a partition scheme), the user would need to tweak the sculpt boot image.
* There are nitpicker-related glitches that need to be fixed:
  * When multiple transparent views are stacked, redraws are sometimes incomplete.
  * The focus-control operations as issued by the window manager interfere with the leitzentrale focus. As the window manager is not a default part of Sculpt EA, this is not a pressing usability issue, but it should be fixed as a potential security - focus stealing - issue.

nfeske, 2018-02-13:
On my sculpt_next branch, I implemented the default block-device idea. So one boot image should cover most laptops now (as long as exactly one AHCI block device is present).

NobodyIII, 2018-02-14:
Can we include part_blk? That's the main component that I found missing when I installed Sculpt.

nfeske, 2018-02-14:
@NobodyIII Sure. The boot module should already be included on my `sculpt_next` branch. So `part_blk` should be usable from `subinit/default_fs.config`.

On request by @skalk, I also put in the hooks required to let a runtime access the other partitions of `part_blk` from the default_fs subinit (he assigns partitions directly to VMs). Maybe @skalk can retry the new version to make sure that I covered everything?

nfeske, 2018-02-14:
Question: Should I better remove the usb_fs runtime? It is not really tested and works only with a few selected USB sticks at best.

nfeske, 2018-02-14:
Furthermore, I'd like to move the nit_focus component from the static system into the runtimes, which allows us to replace it by the window layouter easily.

ehmry, 2018-02-14:
I did not have success with usb_fs, but I appreciate having the *Usb* service available.

nfeske, 2018-02-14:
@ehmry Of course. We need to pass USB sticks to VMs after all.

nfeske, 2018-02-14:
I merged the sculpt scenario (sans documentation) to staging. The following changes are worth noting:

* The "vm" runtime is no more. One has to use the update/deploy runtimes to run VirtualBox.
* The magic keys for the leitzentrale (F11 and F12) can now be configured in the `input_filter.config`.
* The default deploy config uses nitpicker directly (no window manager).
* The depot user is configurable via the RUN_OPT `--depot-user <name>`.
* The nitpicker focus policy moved to the runtime level.

nfeske, 2018-02-14:
Thanks to the support by @alex-ab I was able to successfully download the debian9 installer and install Debian within sculpt. So in principle, we can take all the installation steps without falling back to any 3rd-party Linux live system along the way. Really cool!

However, the download of the 300MiB debian install ISO via fetchurl is unbearably slow. I guess the bottleneck is lwIP, not my network connection. The download speed of the same ISO via wget from within a Linux VM is about 10 times faster. @ehmry how do you think about changing fetchurl to use lxIP instead? Would it be worth giving it a try?

ehmry, 2018-02-15:
@nfeske fetchurl should be using socket_fs, but for me, when removing the lwip plugin the resolver stops working.

nfeske, 2018-02-16:
I updated my sculpt topic branch (https://github.com/nfeske/genode/commits/sculpt) and uploaded all the needed depot content.

Notes for using this version:
* Since I changed the way of how the magic leitzentrale keys are defined, `nitpicker.config` and `input_filter.config` have changed. Of you have customized versions of these files, please review the changes.
* After booting the system, you need to update (e.g., to install virtualbox) via the `update` runtime.
* Once the installation is complete, you may launch the `deploy` runtime to start subsystems out of the runtime-installed packages. 
* The `deploy` runtime starts merely a noux instance by default. Please uncomment start nodes in `config/deploy/config` as needed. For virtualbox, you need to uncomment a nic driver (wifi or ipxe), the nic router, and the vbox section including the vm chroot and shared chroot.

To update and create the sculpt boot image:
```
./tool/depot/create \
    CROSS_DEV_PREFIX=genode-x86- \
    UPDATE_VERSIONS=1 FORCE=1 REBUILD= -j8 \
    genodelabs/pkg/x86_64/sculpt \
    genodelabs/bin/x86_64/base-nova
```

nfeske, 2018-02-16:
I forgot to add the actual step to create the ISO image after creating the sculpt and base-nova packages:

```
make -C build/x86_64 run/sculpt KERNEL=nova
```

nfeske, 2018-02-16:
The boot image will be located at `build/x86_64/var/run/sculpt.iso`.

nfeske, 2018-02-16:
@ehmry By supplying the etc/hosts in addition to etc/resolv.conf, the depot_download scenario works with lxIP. Thanks for making this work! My attempt to move lxIP to a separate VFS server did not succeed though. I pushed the current state to my sculpt_next branch.

ehmry, 2018-02-17:
@nfeske Using a shared stack is fixed at 9468eaec97c5670915aee9bcb5cfbb189f9c3518, more work needs to be done to clean up the noise.

nfeske, 2018-02-19:
@ehmry Fantastic! The depot_download.run scenario succeeds now.

As a minor remark about 9468eaec97c5670915aee9bcb5cfbb189f9c3518, it is better to check the `len` argument via

```
if (len > (sizeof(_content_buffer) - 2))
```

as present in the original code. You changed it to

```
if (len+2 > sizeof(_content_buffer))
```

Even though your version looks more intuitive, it is prone to integer overflows. Let's assume that the function is called with `len = ~0UL`. In this case, `len+2` results in `1` (because of the integer overflow), letting the check pass.

I hope you are fine with me adjusting the patch in this respect.

nfeske, 2018-02-19:
Thanks to @ehmry's move of fetchurl to lxIP, the download performance issue is solved.

I integrated the changes accompanied with several follow-up commits (like a depot recipe for lxip) on both the staging and my sculpt_next branch. @chelmuth I would appreciate a review from your side because I am not very familiar with lxIP.

Working towards the release, our remaining focus should lie on polishing and testing. It would be great of everyone at Genode Labs to switch to the current version of sculpt (of the staging branch) as soon as possible and report/fix remaining issues. Personally, I would not add any non-sculpt-related commit to staging from now until the release to let the depot hashes settle.

Speaking of polishing, in particular the debug noise level of the update runtime is worth improving as there are many false-positive error messages right now. On a related note, it would be nice to add a `verbose` attribute to the fetchurl config. When set to "yes", fetchurl could report the download status every two or so seconds. This would make the downloading of the debian installer (by the `download` runtime) a bit more entertaining ;-)

I will concentrate on the documentation side during the remainder of the week (whenever my time allows).

chelmuth, 2018-02-20:
From a first look into the configuration, it appears to be correct. I'm going to update my system to sculpt_next tomorrow morning (in case you like to add further adaptations).

nfeske, 2018-02-20:
I wanted to rebase sculpt_next to the new staging branch. Unfortunately, lxip does not build anymore. Hopefully, the build error rings a bell for @skalk.

Maybe it would be the best for you to use the staging branch for testing sculpt (my sculpt_next branch differs only in the additional use of a 18px font and the default origin of depot content). To exercise the sculpt-release workflow, you may overtake my role here. You only have to add `--depot-user chelmuth` to your RUN_OPT. For reference, my current sculpt_update.sh script looks as follows:

```
#!/bin/bash

set -x

# force-update pkg and src with versions and build binaries (skipping rebuilds)
./tool/depot/create CROSS_DEV_PREFIX=genode-x86- UPDATE_VERSIONS=1 FORCE=1 REBUILD= -j8 \
	genodelabs/pkg/x86_64/sculpt \
	genodelabs/bin/x86_64/base-nova
	#

# packages for the installation at runtime
./tool/depot/create CROSS_DEV_PREFIX=genode-x86- UPDATE_VERSIONS=1 FORCE=1 REBUILD= -j8 \
	nfeske/pkg/x86_64/sculpt-installation

# generate image
make -C build/x86_64 run/sculpt KERNEL=nova
```

After executing the script and writing the boot image to a USB stick, you also need to publish the pkg/sculpt-installation package. Don't forget to rsync the public/ directory with depot.genode.org. I use the following rsync arguments
```
rsync -rpltOvzu --checksum 
```

chelmuth, 2018-02-22:
After fixing some hassle with the depot web infrastructure, I'm running Sculpt from staging successfully since yesterday afternoon.

nfeske, 2018-02-22:
I'm glad that it works for you! :-)

nfeske, 2018-03-01:
Merged into master.

closed 2018-03-01 by nfeske

ehmry, 2018-03-15:
@nfeske would it makes sense to include the ROM modules for the EGL backend into the Sculpt image? Should the *Gpu* service be provided from drivers? Or should the routing be contained in the deploy runtime? Should packages be released as `foo_swrast` and `foo_i965` instead?

nfeske, 2018-03-16:
I always try to keep software out of the boot image if possible. As EGL is not needed for bootstrapping the system, it should enter the picture not before the deploy level. Therefore, I agree with your latter suggestion.

The GPU multiplexer will become part of the drivers subsystem because it will arbitrate the access to the graphics hardware. In the aspired architecture, the framebuffer driver won't obtain the device directly from the platform driver but from the GPU driver. So the GPU driver is able to keep the GPU-related parts for himself.

That said, the GPU driver is not ready for Sculpt yet. As it clashes with the intel_fb_drv, it can only be used with the VESA driver - which is not selected by Sculpt whenever Intel graphics present.