genodelabs/genode #2677: rump: rump_fs and vfs+rump_fs plugin don't write all data to disk

alex-ab, 2018-02-12:
Using rump_fs server or vfs + rump_fs plugin may produce corrupted data.

I tracked the issue down to missing functionality in the [rump vfs contrib sources](https://github.com/NetBSD/src/blob/trunk/sys/rump/librump/rumpvfs/vm_vfs.c#L67) compared to the [original uvm_pager](https://github.com/NetBSD/src/blob/trunk/sys/uvm/uvm_pager.c#L436) sources of NetBSD. The [error case](https://github.com/NetBSD/src/blob/trunk/sys/uvm/uvm_pager.c#L380) in the orignal uvm pager code is not covered by the [rump vfs contrib sources](https://github.com/NetBSD/src/blob/trunk/sys/rump/librump/rumpvfs/vm_vfs.c#L67).

alex-ab, 2018-02-12:
After adjusting the rump vfs layer accordingly to the orignal uvm_pager code my test case successfully terminates.

nfeske, 2018-02-12:
@alex-ab Wow - you found the needle in the haystack! Hats off for the impressive feat of finding the root cause of this long-standing uncertainty! Thank you very very much for your patience. I merged your fix to staging.

krytarowski, 2018-02-13:
Is this applicable for the upstream sources as well? (In NetBSD's src/).

cnuke, 2018-02-13:
Yes, although it might needs some proper review as it is merely copy of original implementation (I am not sure if rump handles the PG_FAKE case or needs it at all).

krytarowski, 2018-02-13:
We keep developing rump in the NetBSD (mostly for the purpose of tests), best to keep forks in sync as much as possible.

alex-ab, 2018-02-15:
I reported the issue to the [rump kernel user mailing list](https://www.freelists.org/post/rumpkernel-users/Missing-error-handling-in-rumpvfsvm-vfsc) and based on the suggestion by Antti Kantee opened a NetBSD bug report [53032](http://gnats.netbsd.org/53032).

commit 7c5552a4 (2018-02-12): rump: handle error case in vm_vfs rump layer

Fixes #2677
closed 2018-02-16 by nfeske