genodelabs/genode #2678: Dedicated handle type for File_system notifications.

ehmry, 2018-02-12:
This is the first step in my patch series for #1934.

ehmry, 2018-02-12:
I have a newer, cleaner, patch at 63375ee75bacc9fa23bf50f9d52bc6f8703073de. This patch has no effect on the VFS server or library. It is ready for staging, and I am in the process of testing it with sculpt.

nfeske, 2018-02-12:
The patch looks tidy and neat indeed. :-)

When test-driving run/fs_report on base-linux or base-nova, however, the test gets stuck at the 4th step.

ehmry, 2018-02-12:
Fixed at 86dff6491715715367233d4cd8db2c3786c1f255, I was not aware of `run/fs_report`, but I got pretty close anyway.

nfeske, 2018-02-12:
I can confirm it works now. Merged to staging. I am also going to test it in sculpt...

ehmry, 2018-02-12:
I have the same issue with nitpicker, but with none of the other configs.

ehmry, 2018-02-12:
There is an issue here with ``bool Rom_session::update()``.

ehmry, 2018-02-13:
I have the Nitpicker issue fixed, fixup at 3d20debb795624ca840b56af9701ff9b69ebb0a2.

nfeske, 2018-02-13:
Thanks for the update. I'll test drive sculpt with the new version just now.

nfeske, 2018-02-13:
Looks good in Sculpt on Qemu! :-)

nfeske, 2018-02-13:
I merged your fixup to both the genodelabs/staging and nfeske/sculpt_next branch.

nfeske, 2018-03-01:
Fixed in the master branch.

closed 2018-03-01 by nfeske