genodelabs/genode #2679: lxip/vfs: segfault in connect

rite, 2018-02-13:
I tweaked run/netty_udp to use nic_loopback. For KERNEL=nova, vfs crashes with:

```
[init -> netty-client-7070] connected
no RM attachment (READ pf_addr=0xe pf_ip=0x10fc6f1 from pager_object: pd='init -> socket_fs' thread='ep') 
page fault, pd='init -> socket_fs' thread='ep' cpu=0 ip=0x10fc6f1 address=0xe stack pointer=0xa04fef18 qualifiers=0x4 irUwp reason=1
[init -> netty-server-7070] test in blocking mode
```
For KERNEL=linux, the client test finishes 'most of the time', but once in a while a segmentation fault occurs in vfs:

```
[init -> netty-client-7070] connected
[init -> netty-server-7070] test in blocking mode
[init -> socket_fs] Segmentation fault (signum=11), see Linux kernel log for details
```

Here's the stack trace:
```

#0  0x00000000010fc13b in inet_dgram_connect (sock=0x2b133286c460, uaddr=0x2b133a8ff4f0, addr_len=128, flags=0)
    at genode/contrib/dde_linux-6e4a3643875613612c2cc788938aadae5ce7ec7f/src/lib/lxip/net/ipv4/af_inet.c:526
#1  0x0000000001078d2e in Vfs::Lxip_connect_file::write (this=0x2b1332839338, src=<optimized out>, len=13) at genode/repos/dde_linux/src/lib/vfs/lxip/vfs.cc:552
#2  0x00000000010766a2 in Vfs::Lxip_file_system::write (this=<optimized out>, vfs_handle=<optimized out>, src=<optimized out>, count=<optimized out>, out_count=@0x2b133a8ff600: 0)
    at genode/repos/dde_linux/src/lib/vfs/lxip/vfs.cc:1637
#3  0x0000000001001876 in _write (seek_offset=<optimized out>, len=13, src=0x2b13379de710 "10.0.2.1:7070", 'X' <repeats 187 times>..., this=0x2b1337a068a0)
    at genode/repos/os/src/server/vfs/node.h:173
#4  Vfs_server::File::write (this=0x2b1337a068a0, src=0x2b13379de710 "10.0.2.1:7070", 'X' <repeats 187 times>..., len=13, seek_offset=<optimized out>)
    at genode/repos/os/src/server/vfs/node.h:379
#5  0x000000000101772a in operator() (node=..., __closure=<optimized out>) at genode/repos/os/src/server/vfs/main.cc:178
#6  apply<Vfs_server::Node, Vfs_server::Session_component::_process_packet_op(File_system::Packet_descriptor&)::<lambda(Vfs_server::Node&)> > (fn=..., id=..., this=0x2b13379db428)
    at genode/repos/base/include/base/id_space.h:205
#7  _apply<Vfs_server::Session_component::_process_packet_op(File_system::Packet_descriptor&)::<lambda(Vfs_server::Node&)> > (fn=..., handle=..., this=0x2b13379db018)
    at genode/repos/os/src/server/vfs/main.cc:99
#8  Vfs_server::Session_component::_process_packet_op (this=this@entry=0x2b13379db018, packet=...) at genode/repos/os/src/server/vfs/main.cc:179
#9  0x0000000001018216 in _try_process_packet_op (packet=..., this=0x2b13379db018) at genode/repos/os/src/server/vfs/main.cc:225
#10 _process_packet (this=0x2b13379db018) at genode/repos/os/src/server/vfs/main.cc:265
#11 Vfs_server::Session_component::_process_packets (this=0x2b13379db018) at genode/repos/os/src/server/vfs/main.cc:311
#12 0x00002b13326b6c81 in ?? ()
#13 0x00002b13379dd568 in ?? ()
#14 0x0000000000000001 in ?? ()
#15 0xfffffffffffed537 in ?? ()
#16 0x00002b133273e250 in ?? ()
#17 0x0000000000000000 in ?? ()
```

``` 
(gdb) print sock
$3 = (struct socket *) 0x2b133286c460
(gdb) print sock->sk
$4 = (struct sock *) 0x0
```

ehmry, 2018-02-14:
I have a simple fix at 81316e2035af0cac83711d000c8883c6860f5c6a to check that the Linux socket structure is ready for operations.

rite, 2018-02-19:
Altough I don't understand why the socket structure at that point is sometimes (on linux) and always (on nova) not ready, the fix 81316e2 solves my issue, thanks!

commit 7394158c (2018-02-19): vfs_lxip: check that socket is ready for I/O

Check that the Linux socket structure is populated before performing
read or write operations on control files.

Fix #2679
commit 01800ad1 (2018-02-28): vfs_lxip: check that socket is ready for I/O

Check that the Linux socket structure is populated before performing
read or write operations on control files.

Fix #2679
closed 2018-02-28 by chelmuth