genodelabs/genode #2682: Native stub resolver

ehmry, 2018-02-15:
Genode and guests would benefit from a native DNS stub resolver component that would be made available to native network applications and VMs using *nic_router*. Configuring the stub to resolve thru a pool of globally available DNS servers would allow clients to configure themselves with a static `/etc/resolv.conf` to the local stub rather than perpetually update the clients with the nameserver attached to DHCP leases.

m-stein, 2018-02-27:
This is also in my TODO list for the near future.

ehmry, 2018-02-27:
Well my two preferences were [stubby](https://dnsprivacy.org/wiki/display/DP/DNS+Privacy+Daemon+-+Stubby) and [unbound](http://unbound.net/). Stubby is simpler and sort of works out of the box, but does not cache. Unbound caches but does not quite work out of the box.

m-stein, 2018-02-28:
If I'm correct the situation is the following:
Stubby ~2 KLOC, needs porting of getdns lib
Unbound ~10 KLOC, needs openssl lib to work
I would choose stubby for now and if we observe the need for caches consider porting unbound.
I'm working at the NIC router for the next days but after that I could give it a try.

ehmry, 2018-03-01:
I have a test scenario of application lookups via the nic_router and stubby on my **issue2682_stubby** branch at **run/stubby**.

Its just a proof-of-concept though, stubby should be capable of lookups over TCP and TCP-TLS and DNSSEC validation, but I have only had success with using stubby as a proxying to the QEMU nameserver.

ehmry, 2018-03-21:
I have rebased my **issue2682_stubby** branch and added a patch for a nic_router package that provides DNS along with NAT, which I am using now on my laptop.

Stubby DNSSEC / DANE support is still incomplete or untested.

chelmuth, 2018-04-25:
I merged the following series to staging. @ehmry please check if this fits your intention as I did som cleaning of which commit changes which file. Unfortunately, the series will definitely break rebasing.

9999f96 Depot: nic_router-nat-dns
08915e3 Port of getdns and Stubby DNS resolver
49a95ac Port of libyaml
b917ae8 Libc: gracefully handle invalid socketaddr arguments
63bbd9d Test of the gettaddrinfo POSIX facility.

Also, do you see any option to test the feature in the autopilot?

ehmry, 2018-04-25:
Yes, but give me a chance to test the Stubby package with my next sculpt. I want to try out domain blacklisting, then I can make a test for successful and intentionally unsuccessful resolving :-)

ehmry, 2018-06-04:
Fixed in 18.05.

closed 2018-06-04 by ehmry